roboterclubaachen/xpcc #17: extend xpcc::Timestamp to 32bit

daniel-k, 2015-01-29:
At the moment xpcc::Timestamp is of type uint_fast16_t. This limits in particular the usage of xpcc::Timeout to approx. 65 seconds. I have a use case where I need bigger intervals.

The questions is now why 16bit was choosen in particular. As software timeouts and timers are not suitable for time critical applications anyway, I wonder if this could safely be increased to 32bit.

For the moment I can work around this, but having 32bit timestamps would be really nice.

dergraaf, 2015-01-29:
The reason for choosing 16 bit were the limitations of AVR devices when handling 32 bit arithmetic. The generated code for this was considerably larger than for 16 bit and for most actions of our robots the 30 seconds timeout was sufficient. Therefore 16 bit was chosen as a compromise. With 32 bit micro-controllers this limitation does not apply anymore.

Perhaps a good solution would be to make xpcc::Timestamp a template class and use 32 bit as default. This would then also apply to the Timeout and PeriodicTimer classes.

ekiwi, 2015-01-29:
> Perhaps a good solution would be to make xpcc::Timestamp a template class

I basically had the same idea. In order to make this the most user friendly / verbose I'd suggest we create an enum class that indicates the longest possible time span. Then we can use Template specialization to map this to a type. While this would make the xpcc code more complex - I think - it would - at the same time - make the user's code more readable.
Thus something like this:

~~~{.cpp}
xpcc::Timestamp<MaxTime::OneMinute> ts;  // uint16_t
xpcc::Timestamp<MaxTime::FortyNineDays> ts;  // uint32_t
~~~

daniel-k, 2015-01-29:
Ironically my platform is AVR, but I don't care too much for runtime performance in this case.

I also like the idea to have this templatized, but I have a remark on the naming. Having a "static" name like `MaxTime::OneMinute` doesn't work when the clock is supplied with different time bases. AFAIK it's advised by the docs to increment the clock in 1ms intervals, but maybe there are cases where you don't want to or can't have this precision. So having a different time base would change the actual MaxTime value for these constants.

salkinium, 2015-02-08:
> AFAIK it's advised by the docs to increment the clock in 1ms intervals, but maybe there are cases where you don't want to or can't have this precision. So having a different time base would change the actual MaxTime value for these constants.

The `Clock::increment()` method can be given arbitrary numbers to increment the clock value, if your tick interrupt does not have 1ms resolution.
So if your interrupt is only called every 32ms: `xpcc::Clock::increment(32);`. Simpelz.
The timebase is not affected by this.

> Perhaps a good solution would be to make xpcc::Timestamp a template class

Why?
If the maximum resolution of xpcc::Timestamp is uint32_t, then this resolution has to be supported by
xpcc::Clock.
Which means, xpcc::Clock has to always have a uint32_t resolution. Also on AVRs.

Yes, perhaps there is the possibility to improve performance by only checking the lower two bytes of the clock value for `MaxTime::OneMinute`, but your still incrementing a uint32_t in the background every ms.

salkinium, 2015-02-08:
Ok, so I looked at the generated ASM on the AVR8 core of the (modern) ATmega328p for incrementing a uint8_t vs. uint16_t vs. uint32_t value by a constant (of 1).
I annotated the ASM with the instruction description and number of required clock cycles.

My observations:

0. AVRs don't have a ADDI (Add Constant to Register) for uint8_t, they can use SUBI (Subtract Constant from Register) with arithmetic underflow.
1. The time it takes is uint32_t = uint16_t *2 = uint8_t * 4. Nice.
2. The amount of registers used is uint32_t = uint16_t *2 = uint8_t * 4. Not really a surprise.
3. The code size is actually a little better: uint16_t < uint8_t * 2 and uint32_t < uint8_t * 4.
4. The main impact on timing and code size is the moving of the values between SRAM and registers.
The arithmetic operations are not actually that expensive.

I therefore suggest we move forward and give `xpcc::Clock` a uint32_t value to increment, as this is only 10 clock cycles more per increment (or 10.000 cycles more per second at 1ms resolution).

We should however provide two timer implementations as @ekiwi suggested, to optimize the access times and code size of all classes using `xpcc::Timeout`.

@dergraaf
What is the rational behind the template parameter in `xpcc::Timeout`, `xpcc::PeriodicTimer`?
Why/when would you have multiple `xpcc::Clock` instances in your project?

~~~
// volatile uint8_t clock;
// clock++
8a:	80 91 00 01 	lds	r24, 0x0100				// Load Direct from SRAM: 2 clocks
8e:	8f 5f       	subi	r24, 0xFF	; 255	// Subtract Constant from Register: 1 clock (x-255 == x+1 with 8bit underflow)
90:	80 93 00 01 	sts	0x0100, r24				// Store Direct to SRAM: 2 clocks
// Total time: 5 clocks, 4 clocks load/store, 1 clock arithmetic
// Total size: 10B instructions, 8 load/store, 2 arithmetic, + 1 register

// volatile uint16_t clock;
// clock++
8a:	80 91 00 01 	lds	r24, 0x0100				// Load Direct from SRAM: 2 clocks
8e:	90 91 01 01 	lds	r25, 0x0101				// Load Direct from SRAM: 2 clocks
92:	01 96       	adiw	r24, 0x01	; 1		// Add Immediate to Word: 2 clocks
94:	90 93 01 01 	sts	0x0101, r25				// Store Direct to SRAM: 2 clocks
98:	80 93 00 01 	sts	0x0100, r24				// Store Direct to SRAM: 2 clocks
// Total time: 10 clocks, 8 clocks load/store, 2 clocks arithmetic
// Total size: 18B instructions, 16 load/store, 2 arithmetic, + 2 registers

// volatile uint32_t clock;
// clock++
8a:	80 91 00 01 	lds	r24, 0x0100				// Load Direct from SRAM: 2 clocks
8e:	90 91 01 01 	lds	r25, 0x0101				// Load Direct from SRAM: 2 clocks
92:	a0 91 02 01 	lds	r26, 0x0102				// Load Direct from SRAM: 2 clocks
96:	b0 91 03 01 	lds	r27, 0x0103				// Load Direct from SRAM: 2 clocks
9a:	01 96       	adiw	r24, 0x01	; 1		// Add Immediate to Word: 2 clocks
9c:	a1 1d       	adc	r26, r1					// Add with Carry two Registers: 1 clock
9e:	b1 1d       	adc	r27, r1					// Add with Carry two Registers: 1 clock
a0:	80 93 00 01 	sts	0x0100, r24				// Store Direct to SRAM: 2 clocks
a4:	90 93 01 01 	sts	0x0101, r25				// Store Direct to SRAM: 2 clocks
a8:	a0 93 02 01 	sts	0x0102, r26				// Store Direct to SRAM: 2 clocks
ac:	b0 93 03 01 	sts	0x0103, r27				// Store Direct to SRAM: 2 clocks
// Total time: 20 clocks, 16 clocks load/store, 4 clock arithmetic
// Total size: 38B instructions, 32 load/store, 6 arithmetic, +  4 registers
~~~

salkinium, 2015-02-08:
Also, looking at the implementation of `xpcc::Clock`, it already uses `static uint_fast16_t time;` [1].
On 32bit systems this gets promoted to a uint32_t, as it requires no masking on loading/storing and is therefore faster.

So on our 32bit systems, we have already had the benefit of the 32bit clock, we just haven't accessed it yet.

[1]: https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/driver/clock.hpp#L81

dergraaf, 2015-02-08:
@salkinium 
Having the possibility to exchange the clocks is something needed for unit tests. If you test something timing related you need to be able to control the time base in order create useful tests. The same applies to a simulator which simulates the robot code faster than real time.

> So on our 32bit systems, we have already had the benefit of the 32bit clock, we just haven't accessed it yet.

Actually we did. I remember that some of the older robot code depends on the fact that a 32 bit value is used to create longer timeouts.

salkinium, 2015-02-08:
Yes, I just found `xpcc::DummyClock` used for the unittests.

> Actually we did. I remember that some of the older robot code depends on the fact that a 32 bit value is used to create longer timeouts.

[image]

salkinium, 2015-02-08:
I have implemented a potential fix, which does not break the current interface and is therefore backwards compatible.
The default are still 16bit Timestamps for everything (which is enough for most code), but you may explicitly use 32bit Timestamps now, by prefixing `Timestamp`, `Timeout` and `PeriodicTimer` with `Long`.
I think this is the best compromise for the AVR and ARM targets.

If you want to manually access the 32bit timestamp, you need to write a little more now:

~~~{.cpp}
xpcc::LongTimestamp stamp = xpcc::Clock::now<xpcc::LongTimestamp>();
~~~

If this interface is acceptable, I'll merge the feature branch into develop.

salkinium, 2015-02-08:
I found another issue with PeriodicTimer: it's not strictly periodic.
The `isExpired()` method of PeriodicTimer contains this (in short form):

~~~{.cpp}
if (timeout.isExpired()) {
	timeout.restart(interval);
	return true;
}
return false;
~~~

However, this is not periodic, since the `timeout.restart()` restarts from **now**, not from the time it expired:

~~~{.cpp}
`timeout.endTime = Clock::now() + interval;`
~~~

So if `isExpired()` is called too late, and the clock has already moved on a few ms, the PeriodicTimer will simply ignore those missing ms, and the period begins to drift.

I think this is wrong, the code for the PeriodicTimer should be relative to the last expiration:

~~~{.cpp}
if (timeout.isExpired()) {
	timeout.endTime += interval;
	return true;
}
return false;
~~~

However this would make the PeriodicTimer expire in a row for the number of missed periods.
So assume a 10ms PeriodicTimer is delayed by 55ms, then the `isExpired()` method would yield `true` 5 times in a row, until the `timeout.endTime` has caught up with `Clock::now()`.
This is not only confusing, but will very likely yield to unwanted behavior.

The saner alternative is to drop the 4 missed ones, and only expire once out of period, and then be back in sync to the period.
In the above example, the 10ms Periodic Timer would expire only once after 55ms, but schedule itself to expire next at 60ms, instead of 65ms as with the current example.
The code for this period aggregation is less efficient of course:

~~~{.cpp}
Timestamp now = Clock::now() + interval;
do
{
	timeout.endTime = timeout.endTime + interval;
}
while(timeout.endTime < now);
~~~

What do you think?

daniel-k, 2015-02-08:
I guess this only comes to play when you started the timer in sync with something else, otherwise I can't imagine a case where this can be relevant.

On the other hand, with your proposal, isExpired() will yield true in a shorter interval (worst case still 2 times in a row) than it's configured when the last timeout was missed.

So in the end the question may be: Is it better to occasionally have shorter or longer intervals? As PeriodicTimers are not made for time-critical task I'd say leave it as is.

salkinium, 2015-02-09:
Also, my [last commit][1] on the feature branch makes Timeout and PeriodicTimers with 0 (zero) ms period expire on every call, instead of completely stopping the timer.
This is how I would expect this, similar to how `sleep(0);` in usual (non-xpcc) thread classes does not stop the thread.

Was there a specific reason why `Timeout(0)` and `PeriodicTimer(0)` initialized in the stopped state?

[1]: https://github.com/roboterclubaachen/xpcc/commit/8bb1b14b2f838e8fa45ab2e640495162f0e7807a#diff-04c1d0287b399915e0f8ade793e9ad09R23

salkinium, 2015-02-09:
> On the other hand, with your proposal, isExpired() will yield true in a shorter interval (worst case still 2 times in a row) than it's configured when the last timeout was missed.

That is actually a very good point.
Since there is nothing we can do if the expiration is missed, other than expire immediately (which is how we do it now), I agree that this should not lead to a shorter interval afterwards.
Right now we can make the guarantee, that the timeout or period will be **at least** the length specified, but with my solution we cannot actually make *any* guarantee about timing anymore.

So I would agree, that we should leave it as it is now.

salkinium, 2015-02-12:
After much discussion, a new interface for both Timeout and PeriodicTimer has been created.

Here is a summary of the most important changes:

#### Class Naming

`xpcc::Timestamp`, `xpcc::Timeout` and `xpcc::PeriodicTimer` are now 32bit wide for max 24 days at millisecond resolution.
`xpcc::ShortTimestamp`, `xpcc::ShortTimeout` and `xpcc::ShortPeriodicTimer` are 16bit wide for max 32s at millisecond resolution.

These classes are complete specializations of `xpcc::GenericTimestamp<Type>`, `xpcc::GenericTimeout<Clock, TimestampType>` and `xpcc::GenericPeriodicTimer<Clock, TimestampType>` for `uint_fast16_t` and `uint32_t` and `xpcc::Clock` and `xpcc::ShortTimestamp`/`xpcc::Timestamp` respectively.

So there will be no more redundant `<>` after `Timeout` or `PeriodicTimer`. Use `Generic*` if you need a special clock.

#### Misunderstanding isExpired() 

`isExpired()` behaved and was used differently for both Timeout and PeriodicTimer.

In Timeout, it kept its state after expiration, as it should, which made it difficult to execute code only once after expiration.
A hacked solution would be `if (timeout.isActive && timeout.isExpired())`, which would trigger only once after expiration.

In PeriodicTimer, `isExpired()` expired only once, which was great for executing code a maximum of once per period, but did not make any sense when comparing the method's behavior to the same method in Timeout.

We decided to decouple the Timeout/PeriodicTimer state logic from code execution.
Both classes receive a `execute()` methods, which returns `true` only once after expiration and per period respectively.

Additionally a side-effect-free `getState()` method with the associated enum class would allow querying of state logic at any given time.

The Timeout class, still retains its `isExpired()` method, which is a convenience function for `getState == TimeoutState::Expired`, but functions exactly as before.

This means that the Timeout class can be used without calling `execute()` as it is usually done for waiting in a lot of Protothreads (we really ought to consider a form of `PT_SLEEP(time)` for convenience).
However, a PeriodicTimer should not be used in this way.

#### Example Code

~~~{.cpp}
xpcc::Timeout timeout(1000);
xpcc::PeriodicTimer timer(250);

while(1)
{
    if (timeout.execute()) {
        Led1::toggle();    // executed only once after 1s
    }
    if (timeout.isExpired()) {
        Led2::toggle();    // executed continuously after 1s
    }
    if (timer.execute()) {
        Led3::toggle();    // executed once every 250ms
    }
}
~~~

What do you think?

Unfortunately refactoring `isExpired()` to `execute()` for PeriodicTimers has to be done manually, as there is no simple way of distinguishing Timeouts from PeriodicTimers by object name.
There seems to be a trend to name Timeout objects `timer`, when they should really be named `timeout`.

commit fc5ebd17 (2015-02-10): Merge branch 'feature/extended_timestamp' into develop. Fixes #17.
closed 2015-02-12 by salkinium