roboterclubaachen/xpcc #28: Use Cortex-M3/M4 Bit-Banding for atomic access

salkinium, 2015-08-14:
The GPIO operations are not atomic, due to being a Read-Modify-Write operation!
They are also slower than they should be.

We should access the Bit-Banding Memory region to perform such time critical operation, which are guaranteed to be atomic by the hardware itself.

Here is a short introduction to the topic of Bit-Banding.
http://spin.atomicobject.com/2013/02/08/bit-banding/

salkinium, 2015-08-14:
The issue with the GPIO in particular is that for `SoftwareI2cMaster` there is quite a delay only from reading and writing the GPIOs. This reduced the resulting datarate quite considerably.

The bit-banging implementations are limited due to the use of the µs delay functions and since the I2C implementation requires four delays per clock, this resulted in 4µs minimum clock (ie. 250kHz).
However, I believe I measured something closer to 200kHz as maximum clock on the STM32F4, due to the additional code execution and also GPIO access times.

That's not so bad for the slow I2C bus, but for `SoftwareSpiMaster`, this topped out at about 0.4MHz, which is simply too slow, even for a bit-banging implementation.

There is a `_delay_ns` implementation available on the `feature/cycle_counter` branch, however, this has a minimum delay of 172ns, which is ~5MHz with calling overhead and I would like the influence of GPIO access to be reduced as well.

ekiwi, 2015-08-15:
Aren't the `set`, `reset`, `setOutput` methods already atomic?

The template source for the `set` method on stm32 for example looks like this:

~~~{.cpp}
	ALWAYS_INLINE static void set() {
			%% if target is stm32f3
		{{reg}}->BSRRL = mask;
			%% else
		{{reg}}->BSRR = mask;
			%% endif
	}
~~~

Source: https://github.com/roboterclubaachen/xpcc/blob/feature/cycle_counter/src/xpcc/architecture/platform/driver/gpio/stm32/gpio.hpp.in#L233

salkinium, 2015-08-15:
Hm, that is a very solid point. I think the offset didn't come from the GPIO, but from the inaccurate `_delay_us` function, which was about 10% too slow, which would explain the difference.

closed 2015-08-15 by salkinium