roboterclubaachen/xpcc #32: Potential race condition in LPC11C24 uart

porkybrain, 2015-09-09:
in: https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/uart/lpc/uart.cpp.in#L328 LPC_UART->IER is accessed in a non atomic manor. Even though this is an interrupt handler the user could still be calling read/write/discardTransmitBuffer... from a higher priority interrupt.

salkinium, 2015-09-09:
That's true, a `xpcc::atomic::Lock lock;` placed into the if-branch would solve that.
On the Cortex-M3 and up you could use Bit-Banding to archive atomicity in hardware without any overhead.

However, **nothing in xpcc is re-entrant**! You must not call any API from an interrupt context unless specifically declared that you can!
So this is true, but there are a lot of other places where you will find "errors" like these, because you are not meant to do stuff in interrupts in xpcc.

Unfortunately the LPC11 support in xpcc has not seen much love, because quite frankly the hardware is a little annoying to implement for. The GPIOs are not consistent in memory mapping, the UART FIFO is missing a "FIFO is full" flag and the documentation in general could be better.
We've actually only used the LPC11C24 for its integrated CAN controller, but moved away due to its limitations, so a lot of the support has fallen behind.

Would you like to fix this?
I don't have the best overview of the LPC series, I'm more in touch with the STM32 stuff, because they have machine readable memory maps and peripheral configurations, which is a lot more fun to parse and create little helper tools for ;-)

porkybrain, 2015-09-09:
[xpcc not reentrant] Ah ok I actually figured as much after reviewing other code.
[fixing LPC11] I have used a LPC11U68 in a design and we are considering using an LPC11C24 soon. If I find time I would be glad to help.

closed 2015-09-10 by salkinium