roboterclubaachen/xpcc #34: Suggest naming prefixing all unsafe member functions with unsafe

porkybrain, 2015-09-11:
Here https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/driver/atomic/container.hpp#L158 atomic::Container provides an unsafe function, which is perfectly usefull and ledgitimate. However there is a convention of prefixing all function which violate assumed constraints with 'unsafe' as in `unsafeDirectAccess`. That way the user or code reviewer knows when they must read the documentation and when they can just assume the best. I would suggest using this convention in xpcc. (this is more of a topic for a mailing list but I did not find one here http://xpcc.io/sitemap.html, I have been known to be stupid though ;)

ekiwi, 2015-09-11:
> This is more of a topic for a mailing list but I did not find one here http://xpcc.io/sitemap.html, I have been known to be stupid though ;)

There is a mailing list, but it's hidden at the bottom of our [Readme](https://github.com/roboterclubaachen/xpcc/blob/develop/README.md#how-can-i-contribute) file.

Thank you very much for your code review by the way and sorry, that we aren't as quick to respond. Niklas has just started his new job, I'm in the final month(s) of my Bachelor's thesis and most of the code that you have commented on has been in xpcc before we even started working on it, so we aren't to familiar with it either. But we will get to it eventually.
Thanks again!

porkybrain, 2015-09-11:
You guys are actually super responsive compared to Niklas's new employer ;) check this out https://github.com/mbedmicro/mbed/issues/1056 now one seems to care since august. Thats why I'm reviewing your code rather than theirs ;)

0xc0170, 2015-09-11:
@porkybrain You should not put all ppl into the same "bag" :) Note taken. If I respond, will you resume your review? :grinning:

porkybrain, 2015-09-11:
Hey Martin, I know your one of the good ones! I have heard from many sources that there is a major update coming for mbed which changes threading dynamics. When it comes out I will give it another whirl.