roboterclubaachen/xpcc #35: Rename xpcc "SmartPointer"

ekiwi, 2015-09-30:
So I just stumbled upon the xpcc `xpcc::SmartPointer` which confused me for a while, until I had a look at the implementation. While it might not have been the case in 2009, currently a [smart pointer](https://en.wikipedia.org/wiki/Smart_pointer) is a pretty specific concept used in the c++ standard library.
I just assumed, that a `xpcc::SmartPointer` would act the same as the `std::shared_ptr` which it does not. I figured that out after searching for the point at which memory to be managed by the `xpcc::SmartPointer` was allocated. This point does not exist, as the `xpcc::SmartPointer` exclusively manages a runtime sized `uint8_t` array which it [allocates internally](https://github.com/roboterclubaachen/xpcc/blob/a0ff6fc7710135b457ddf211b6447c9dad9144da/src/xpcc/container/smart_pointer.cpp#L53).
Thus it makes sense to keep the `xpcc::SmartPointer` implementation in the `container` directory. 

The naming however is imho very misleading.

I therefore propose the rename `xpcc::SmartPointer`. I haven't come up with a good enough name yet, but some possibilities are:
* `xpcc::SharedMemory`
* `xpcc::ManagedMemory`
* `xpcc::ManagedSharedMemory`

Does anyone have a better suggestion, or does anyone oppose a name change?

salkinium, 2015-09-30:
+1. I was also extremely confused by the naming of this thing, and I very much dislike its internal implementation.
`std::shared_ptr` alongside `std::make_shared` are the only acceptable smart pointer for wide use.
Of course, we would have to port it to AVR, since we currently have no Standard Library (also see #20).

If I recall correctly, the `xpcc::SmartPointer` is not templated for a type, so that it can be passed to the `xpcc::Dispatcher`. It was meant as an internal XPCC mechanism to keep track of memory, and is not meant for wider use in the framework.

mredd, 2015-09-30:
> Am 30.09.2015 um 20:34 schrieb Kevin Laeufer <notifications@github.com>: 
> I therefore propose the rename xpcc::SmartPointer. I haven't come up with a good enough name yet, but some possibilities are:
> 
> xpcc::SharedMemory
> xpcc::ManagedMemory
> xpcc::ManagedSharedMemory
> Does anyone have a better suggestion, or does anyone oppose a name change?
> 

I haven't had a Look into the exact implementation, but the concept you described reminds on what boost is calling scoped_array<uint8_t> for many, many years.

http://www.boost.org/doc/libs/1_59_0/libs/smart_ptr/scoped_array.htm

If the xpcc implementation features reference counting, you probably want to call it shared_array.

Regards,
Christoph

salkinium, 2015-09-30:
Oh and btw: `xpcc::SmartPointer` can actually be defined as a `build-in` type for XPCC communication!
```xml
<builtin name="xpcc::SmartPointer" size="4" />
<typedef name="Data Pointer" type="xpcc::SmartPointer"/>
```
(size must be 2 for AVRs).
The support for that was secretly enabled by me in the innocent looking commit 6f2dfa9 ;-)

This was a hack to allow internal sharing of large amounts of memory safely without copying it.
This obviously only works for a local event loop, something that isn't currently supported very well, but can be implemented using an empty implementation of `xpcc::BackendInterface`.