roboterclubaachen/xpcc #36: xpcc unittests: fix output for function pointers

ekiwi, 2015-10-09:
I have a unittest, that compares two function pointers. They point to different functions.

I get the following output:

~~~
FAIL: message_parser_test:154 : true == true
~~~

Which is somewhat confusion.

strongly-typed, 2016-02-29:
Please take a look at

https://github.com/strongly-typed/xpcc/tree/feature/printf_fptr

for some improvements. At the moment the unit test is made to fail to show the behaviour of the new `<<` operator. We need unit testing for unit tests ;-)

# STM32 target

     scons unittest target=stm32
     arm-none-eabi-nm /Users/usr/Dev/xpcc/build/unittest_stm32/executable.elf -S -C --size-sort -td
     [...]
     134274424 00000006 T myFunc1()
     134274216 00000006 T myFunc2()
     [...]

So calculating the hex of the addresses:

     134274424 = 0x800DD78
     134274216 = 0x800DCA8

The runtime result on an STM32F4Discovery board is just off by one :-O

     Unittests (Feb 29 2016, 17:49:17)
     [...]
     FAIL: io_stream_test:533 : 0800DD79 == 0800DCA9

# OS X 10.11

     Linking: /Users/usr/Dev/xpcc/build/unittest_hosted/executable
     FAIL: io_stream_test:533 : 0000000106871998 == 00000001068719A3

Any suggestions?

salkinium, 2016-09-14:
Would you like to add this solution as a PR?

strongly-typed, 2016-09-15:
Here you go #186. It may still be off-by-one on an STM32.

strongly-typed, 2016-09-15:
I will check if the other features of my branch are still applicable and will then open another PR(s).

strongly-typed, 2016-09-15:
All the other features of that branch seem to be already integrated into RCA/develop. So this issue can be closed and I will then delete my feature branch.

daniel-k, 2016-09-15:
In case you're still wondering about the off-by-one: that's because these are function pointers to Thumb instructions. See [this](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489e/Cihfddaf.html)

ekiwi, 2016-09-15:
> function pointers to Thumb instructions

shouldn't thumb instructions be 16bit aligned?

salkinium, 2016-09-15:
Sigh. Thumb vs. ARM mode. So much confusion.

daniel-k, 2016-09-15:
> shouldn't thumb instructions be 16bit aligned?

@ekiwi 
Sure, that's why the LSB can be used to tell the processor which kind of instruction (ARM vs Thumb) it's going to branch to.

closed 2016-09-15 by salkinium

ekiwi, 2016-09-15:
> Sure, that's why the LSB can be used to tell the processor which kind of instruction (ARM vs Thumb) it's going to branch to.

Thank you, that was the point that I was missing.

salkinium, 2016-09-15:
The compiler is quite smart to hide it in normal operation, but it's a PITA if you forget it if you do some manual function pointer arithmetic in ASM.

PR #186 (2016-09-15): [io] IOStream: Add a << operator for function pointers.
Rebased and polished as requested. Closes #36.