roboterclubaachen/xpcc #50: BitOperation: bitReverse does not work on cortex-m0

ekiwi, 2016-01-07:
Apparently the [inline assembly](https://github.com/roboterclubaachen/xpcc/blob/3aaf58192c005ddea2478dc51c7faa2071f7746d/src/xpcc/math/utils/bit_operation.hpp#L171) used in the `bitReverse` function does not work on cortex-m0.

I get the following error from the compiler:

~~~
/tmp/ccykPtaf.s: Assembler messages:
/tmp/ccykPtaf.s:194: Error: selected processor does not support Thumb mode `rbit r1,r1'
/tmp/ccykPtaf.s:261: Error: selected processor does not support Thumb mode `rbit r3,r3'
~~~

salkinium, 2016-01-07:
Hm, that's because the `RBIT` instruction isn't part of the Cortex-M0 instruction set, only Cortex-M3 and up have it.
[image]

ekiwi, 2016-01-07:
Ok. Is there a good way to detect the cortex-m0 in order to employ the fallback code?
CMSIS uses `__CORTEX_M`, but this seems to be set in the corresponding headerfile of the chip, that is not available (or: that should not be used) outside of `architecture/platform`.

salkinium, 2016-01-07:
Yes, it already uses `detect.hpp`, so just use the `XPCC__CPU_CORTEX_M0` macro.

PR #51 (2016-01-08): [math] BitOperations: rbit operation is not available on cortex-m0
fixes #50
commit 4caefdbf (2016-01-08): [math] BitOperations: rbit operation is not available on cortex-m0

fixes #50
closed 2016-01-08 by salkinium