roboterclubaachen/xpcc #52: CMSIS:  warning: "__CORTEX_SC" is not defined

ekiwi, 2016-01-08:
When compiling for cortex-m0 I get the following warning from CMSIS header files:

~~~
xpcc/ext/cmsis/stm32/Include/core_cmInstr.h:541:36: warning: "__CORTEX_SC" is not defined [-Wundef]
 #if       (__CORTEX_M >= 0x03) || (__CORTEX_SC >= 300)
~~~

`__CORTEX_SC` apparently refers to the [ARM SecureCore Processors](http://www.arm.com/products/processors/securcore/) that are also supported by the CMSIS, but not used by us.
`__CORTEX_SC` contains the SecureCore revision when either [core_sc000.h](https://github.com/roboterclubaachen/xpcc/blob/3aaf58192c005ddea2478dc51c7faa2071f7746d/ext/cmsis/stm32/Include/core_sc000.h#L76) or [core_sc300.h](https://github.com/roboterclubaachen/xpcc/blob/3aaf58192c005ddea2478dc51c7faa2071f7746d/ext/cmsis/stm32/Include/core_sc300.h#L76) are included.
In `core_sc000.h`, `__CORTEX_SC` is defined to be `0`, thus just defining it as zero won't help us and won't make semantic sense.

The correct fix (TM) would - imho - be to check if `__CORTEX_SC` is defined and then check its value:

~~~{.c}
 #if       (__CORTEX_M >= 0x03) || (defined(__CORTEX_SC) && (__CORTEX_SC >= 300))
~~~

However, I do not really want to touch the CMSIS library. @salkinium , can you poke some people at ARM to maybe fix this? [ChibiOS](http://forum.chibios.org/phpbb/viewtopic.php?f=16&t=2740) has noticed the same problem, so I'm pretty sure it should be fixed in CMSIS.

ekiwi, 2016-01-08:
Btw: the reason, this does not show up with other cortex-m versions is, that the first condition is true (`__CORTEX_M >= 0x03`) and thus the second condition is not evaluated.

ekiwi, 2016-01-08:
Should we update to [CMSIS 4.5](https://github.com/ARM-software/CMSIS)?
Maybe even as a submodule?

The bug described here is not fixed, but if we use the latest version from ARM, we could file a bug on their github repo.

salkinium, 2016-01-08:
I wouldn't put it as submodule in xpcc, since we only need very little of it (really only CMSIS/Include* and CMSIS/DSP_Lib/Source/*).
We can fork the CMSIS repo and file a PR with a fix and then we'll see what happens.

0xc0170, 2016-01-11:
@ekiwi please create an issue in https://github.com/ARM-software/CMSIS/, or even send if upstream

ekiwi, 2016-01-11:
@0xc0170 thank you for the suggestion: I have created a pull request: https://github.com/ARM-software/CMSIS/pull/18

salkinium, 2016-02-09:
We should apply the bug fix to our own code.
I asked (Keil to clarify the contribution process](https://github.com/ARM-software/CMSIS/issues/1#issuecomment-181139259), but I'm not sure they have any resources allocated to reviewing outside contributions.

salkinium, 2016-02-13:
Fixed in https://github.com/roboterclubaachen/xpcc/commit/cac3cb7240fe35aeefd4b90882d0683c1c3ec021.

closed 2016-02-13 by salkinium