roboterclubaachen/xpcc #58: use F_CPU to initialize xpcc::clock::fcpu_MHz and xpcc::clock::fcpu_kHz

ekiwi, 2016-01-09:
`F_CPU` is generated from the clock setting in the `project.cfg`.
Is there any reason not to use this value, to initialize `xpcc::clock::fcpu_MHz` and `xpcc::clock::fcpu_kHz`?

salkinium, 2016-01-09:
I wanted to purge `F_CPU` (and `clock` in the `project.cfg`), since it's declaration is redundant and error prone.
The `SystemClock` class knows the final CPU speed best and writes that information into the global variables.

I know that the 8MHz is not always accurate, ie. some STM32 come with a 16MHz internal clock, but `F_CPU` is really the wrong semantics to use it like that.

ekiwi, 2016-01-09:
Don't we need to inject the CPU frequency for AVR microcontrollers?

salkinium, 2016-01-09:
Yes, of course only remove `clock` and `F_CPU` for Cortex-M devices (and perhaps Xmega devices).

(It might be possible to do some tricks though to get the same SystemClock interface for AVRs working, which only generates the Fuse Bit Values and places them inside a specific linker section.
This could then be interpreted by the toolchain to flash it automatically with avrdude.)

PR #92 (2016-02-13): Initialize xpcc::clock::fcpu frequencies correctly.
Add data about internal clock frequencies of the STM32 family and LPC11 and initialize variables separately.
This feature also enables clock driver support for STM32F0 incl. HSI48 and HSI14 support.
I rewrote the example with correct flash latency support.

Fixes #58.
closed 2016-02-13 by salkinium