roboterclubaachen/xpcc #65: xpcc::delayMilliseconds() inaccurate

rleh, 2016-01-26:
`xpcc::delayMilliseconds(50)` on a STM32F072 (cortex-m0) using internal (HSI) or external (HSE) clock (8MHz) delays about 66ms.

salkinium, 2016-01-26:
Thanks for measuring that!

On Cortex-M3 and higher `xpcc::delay*seconds()` uses the Cycle Counter of the Debug module, which give you an absolute cycle accuracy of a few cycles, irrelevant of delay time.

On the Cortex-M0 the cycle counter is not available, so this is [implemented using approximative cycle counting](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/delay.cpp.in#L75-L80), which is obviously not an exact science.

Here the 66/50, aka. 4/3 split seems to indicate that the delay loop takes one cycle longer than I thought.
Let me see if I can fine tune the delay loop and verify correct timings.
(You can try this yourself [by setting the `loop` constant here to 4.](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/delay.cpp.in#L13))

In the mean time I recommend waiting for an `xpcc::Timeout`, which uses the SysTickTimer, which should be a lot more accurate.
```cpp
xpcc::Timeout timeout(50);
while(not timeout.isExpired()) ;
```
I actually recommend this approach for anything above 1ms, as the delay functions are meant for less than that (often required for latching delay of a bit banged bus).

salkinium, 2016-01-26:
So, I can reproduce this and the timing is fixed by setting `loop` to 4.

salkinium, 2016-01-26:
A little heads up: The `xpcc::delayNanoseconds()` method will not be accurate at 8MHz on Cortex-M0, since the `xpcc::clock::ns_per_loop` value is not correctly initialized for a 4-cycle delay loop.
That won't be fixed so easily however.

PR #66 (2016-01-26): Fix delay loop timing for STM32F072 / Cortex-M0
The delay loop on the Cortex-M0 requires 4 cycles, not 3.
Fixes #65.
closed 2016-01-26 by salkinium