roboterclubaachen/xpcc #67: QTCreator Support

ekiwi, 2016-01-27:
@rleh and I started playing around with [QTCreator](https://www.qt.io/ide/) yesterday.

This issue will track our progress in integrating `xpcc` with QTCreator

ekiwi, 2016-01-27:
We basically used the `Existing Project` template as documented in the [scons wiki](https://bitbucket.org/scons/scons/wiki/IDEIntegration)
We used `scons verbose=1` to see the compiler flags and then manually copied all `-D` flags to the `.config` file and  all `-I` flags to the `.includes` file. This can probably be easily automated with `scons`.
This already gives you decent auto completion and inspection (`F2`) support.

ekiwi, 2016-01-27:
QTCreator has some defines set by default, which lead the `detect.hpp` from `xpcc` to detect a hosted linux target. Thus some stuff that uses this result leads to wrong inspection results. For example the hosted implementation of `delayMilliseconds` was resolved, even though we were targeting the ARM platform.
@rleh has found out a way to work around this issue by using some additional `#define` and `#undef` in the `.config` file.

ekiwi, 2016-01-27:
Another way to resolve the above mentioned issue might be to get rid of the `detect.hpp` magic, as it is probably compiler specific and as we know the platform anyway from the device file and could propagate that information through compiler flags or through a auto generated `defines.hpp`.

rleh, 2016-01-27:
The following `#define` and `#undef` in the `.config` file lead `detect.hpp` detect an ARM platform:
```
#define XPCC__CPU_ARM
#undef XPCC__OS_UNIX
#undef __unix__
```

salkinium, 2016-01-27:
Probably not helpful for the `detect.hpp` issue, but there exists also the [QT Bare Metal plugin](http://doc.qt.io/qtcreator/creator-developing-baremetal.html). This was also presented at FOSDEM the last years.

hb128, 2016-02-24:
Thanks for your nice work :)
Could .xml files be added by default as well?

ekiwi, 2016-02-24:
> Could .xml files be added by default as well?

Unfortunately, generating the xml files is not that easy.
The `*.user` file contains user specific settings and should never be automatically overwritten which is why I do not want to generate it.
On the other hand, one should be able to creator a `.shared` file, as is briefly explained in the [QtCreator docs](http://doc.qt.io/qtcreator/creator-sharing-project-settings.html).
However, I haven't figured out how to create a `.shared` file that contains the build commands and that is accepted by QtCreator.

hb128, 2016-02-25:
Sorry, I was a bit ambigous: My intention was not to include the xml files used by qtcreator, but the xml files for the the xpcc communication (components, events, packets) in our robot's project.