roboterclubaachen/xpcc #69: 'round' is not a member of 'std' on Ubuntu 14.04

salkinium, 2016-02-02:
Compiling `example/stm32f4_discovery/blink` inside the xpcc-vm leads to the following compile error:
```
In file included from /home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/drivers.hpp:67:0,
                 from /home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/platform.hpp:25,
                 from ../stm32f4_discovery.hpp:19,
                 from main.cpp:1:
/home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/clock/stm32/static.hpp: In static member function 'static xpcc::clock::StartupError xpcc::stm32::PllSetup<InputFrequency, OutputFrequency, UsbFrequency, Source>::enable()':
/home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/clock/stm32/static.hpp:67:32: error: 'round' is not a member of 'std'
     xpcc::clock::ns_per_loop = std::round(1000000000.f / OutputFrequency * 3.f);
                                ^
/home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/clock/stm32/static.hpp:67:32: note: suggested alternative:
In file included from /usr/arm-none-eabi/include/c++/4.9.3/cmath:44:0,
                 from /home/rca/xpcc/src/xpcc/math/filter/pid.hpp:35,
                 from /home/rca/xpcc/src/xpcc/math/filter.hpp:41,
                 from /home/rca/xpcc/src/xpcc/math.hpp:39,
                 from /home/rca/xpcc/src/xpcc/math/tolerance.hpp:14,
                 from /home/rca/xpcc/src/xpcc/architecture/interface.hpp:15,
                 from /home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/can/generic/can_bit_timings.hpp:13,
                 from /home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/drivers.hpp:36,
                 from /home/rca/xpcc/build/stm32f4_discovery/blink/libxpcc/src/xpcc/architecture/platform.hpp:25,
                 from ../stm32f4_discovery.hpp:19,
                 from main.cpp:1:
/usr/arm-none-eabi/include/math.h:278:15: note:   'round'
 extern double round _PARAMS((double));
```

ekiwi, 2016-02-03:
any idea, why this builds on travis?

salkinium, 2016-02-09:
I have to check if I have original version of the `xpcc-vm`. I might have misconfigured mine.
We provide [the "std" lib shim for this here.](https://github.com/roboterclubaachen/xpcc/blob/develop/src/stdc%2B%2B/cmath#L103), but probably only for double, not for float.

Thinking about this more, `::round` is very likely not a `constexpr` function, so this is not what I want anyways. We ought to use the compilers native C++ stdlib for ARM, and only rolling our own for AVR.

salkinium, 2016-02-20:
I couldn't reproduce this anymore.

dergraaf, 2016-03-06:
Other (including me) are also running into that problem:
https://answers.launchpad.net/gcc-arm-embedded/+question/264485

I'm using the default arm-none-eabi-gcc (and corresponding libstdc++ and newlib) versions of Xubuntu 15.10 (arm-none-eabi-g++ version 4.9.3 20150529).

Apparently _GLIBCXX_USE_C99_MATH_TR1 is not set which disables all standard math functions in the std namespace. Setting it manually results in the following error:
```
In file included from main.cpp:3:0:
/usr/include/newlib/c++/4.9/cmath:1120:11: error: '::log2l' has not been declared
   using ::log2l;
           ^
/usr/include/newlib/c++/4.9/cmath:1124:11: error: '::logbl' has not been declared
   using ::logbl;
           ^
/usr/include/newlib/c++/4.9/cmath:1146:11: error: '::nexttoward' has not been declared
   using ::nexttoward;
           ^
/usr/include/newlib/c++/4.9/cmath:1147:11: error: '::nexttowardf' has not been declared
   using ::nexttowardf;
           ^
/usr/include/newlib/c++/4.9/cmath:1148:11: error: '::nexttowardl' has not been declared
   using ::nexttowardl;
```

It appears that newlib is missing the implementation for these functions which then in terms disables the std versions in C++11 altogether :-(

salkinium, 2016-03-06:
Would it be possible to set `_GLIBCXX_USE_C99_MATH_TR1` in the xpcc build system manually as a work around?

dergraaf, 2016-03-06:
No, as I said this triggers the error above.
The only solution is to use ::round (without std) or replace the installed stdlibc++.

salkinium, 2016-03-06:
> No, as I said this triggers the error above.

Sry, I missed that part.

Could we create a work around for this [like for the AVRs](https://github.com/roboterclubaachen/xpcc/blob/develop/src/stdc%2B%2B/cmath#L103)?

ekiwi, 2016-03-06:
Are we sure that this is not a bug in the Ubuntu package?
`std::round` seems to work well with the newlib from launchpad as well as the one from the Fedora package repository.

dergraaf, 2016-03-06:
> Are we sure that this is not a bug in the Ubuntu package?

It might be. I can check it against the Debian 8 package tomorrow.
Nonetheless is there any harm in using ::round?

salkinium, 2016-03-06:
> Nonetheless is there any harm in using ::round?

Not really. I was kinda hoping this would be optimized away into a constant. I assumed `std::round()` would be a `constexpr` function, but I might have been wrong.

dergraaf, 2016-03-06:
> I was kinda hoping this would be optimized away into a constant.

This appears to happening anyway, even without it being a constexpr function. GCC does a lot of optimization on the standard math functions. Going through the generated Assembler Code I couldn't find any floating point operations:

```
				xpcc::clock::ns_per_loop = round(1000000000.f / OutputFrequency * 3.f);
 8000212:	4b11      	ldr	r3, [pc, #68]
 8000214:	2212      	movs	r2, #18
 8000216:	801a      	strh	r2, [r3, #0]
```

This loads the constant 18. This is the expected result for `round(17.857)` (OutputFrequency is 168000000).

salkinium, 2016-03-06:
Ok, cool, want to send a PR?

This is not the only `std::round` in this area ;-)
```
examples/stm32f072_discovery/stm32f072_discovery.hpp
52:		xpcc::clock::ns_per_loop = std::round(4000 / (Frequency / 1000000));

examples/stm32f7_discovery/stm32f7_discovery.hpp
97:		xpcc::clock::ns_per_loop = std::round(1000 / (Frequency / 1000000));

src/xpcc/architecture/platform/driver/clock/lpc/static.hpp.in
65:				xpcc::clock::ns_per_loop = std::round(1000000000.f / OutputFrequency * 4.f);

src/xpcc/architecture/platform/driver/clock/stm32/static.hpp.in
73:				xpcc::clock::ns_per_loop = std::round(1000000000.f / OutputFrequency * 3.f);
117:				xpcc::clock::ns_per_loop = std::round(1000000000.f / OutputFrequency * 3.f);
```

dergraaf, 2016-03-06:
> Ok, cool, want to send a PR?

Can do. I will do so tomorrow.

dergraaf, 2016-03-07:
> I can check it against the Debian 8 package tomorrow.

Testing it on Debian 8 machine with the standard arm-none-eabi package yields the same result as the Xubuntu version, no `std::round` available.

PR #106 (2016-03-07): Use ::round instead of std::round.
The round function is not available in the std namespace on all ARM libstdc++ implemenations. Fixes #69
closed 2016-03-07 by salkinium