roboterclubaachen/xpcc #81: port hardfaulthandler to cortex-m0

ekiwi, 2016-02-08:
I am currently debugging a potential hardfault on cortex-m0 and it would be helpful, if our awesome handler code would be available.

Maybe @daniel-k can look into this issue, as he has gained some insight into handling hardfaults on a cortex-m0 based device during his Bachelor's thesis.

ekiwi, 2016-02-08:
nevermind (#38)

closed 2016-02-08 by ekiwi

salkinium, 2016-02-08:
It would still be nice if @daniel-k can have a look at the current STM32F0 implementation (this commit https://github.com/salkinium/xpcc/commit/ef8c2991c8d35f70a0f26b9f47c05c34727f73a6).
In my tests, registers r0-r3 were not copied correctly, even though they should have been.
The values in the stack printout were correct however…

daniel-k, 2016-02-08:
I will need a board to step through the code to get a better understanding. I already found some  magical numbers that will need documentation, but I'll point that out again when testing the code.

salkinium, 2016-02-20:
ping @daniel-k: magic numbers?

daniel-k, 2016-02-20:
Totally forgot about this, sorry. Concerning the magic number: that was probably when I looked at the whole hardfault handler. The linked commit looks fine on the first sight. Still didn't test though ...