roboterclubaachen/xpcc #83: Cortex-M0 Startup: vector_table_in_ram broken

ekiwi, 2016-02-08:
It is not enough to move the vector table to ram, the NVIC controller also needs to be notified.
Unfortunately, the Cortex-M0 NVIC does not provide the `VTOR` register, thus a manufacturer specific solution needs to be used.

Some examples for this can be found in the mbed repository:
* [STM32F0](https://github.com/mbedmicro/mbed/blob/4f5900646773ccd5138fb409234d97f1cd67ea98/libraries/mbed/targets/cmsis/TARGET_STM/TARGET_STM32F0/TARGET_NUCLEO_F072RB/cmsis_nvic.c#L46)
* [LPC11xx](https://github.com/mbedmicro/mbed/blob/4f5900646773ccd5138fb409234d97f1cd67ea98/libraries/mbed/targets/cmsis/TARGET_NXP/TARGET_LPC11XX_11CXX/cmsis_nvic.c#L34)

Thus moving interrupt vectors to RAM on `cortex-m0` targets is currently broken.

salkinium, 2016-02-08:
Do we have a concept for moving the vector table to RAM?
There currently is no abstraction, so we would need to implement `NVIC_SetVector` or similar as well.

Keep in mind that we cannot change the peripheral interfaces, since you cannot have a dynamic vector table on the AVR and therefore not include this in the HAL design.
Maybe as a platform specific interface extension? It does make a lot of sense on the Cortex-M.

ekiwi, 2016-02-09:
> Do we have a use case for moving the vector table to RAM?

Not really.

I just wanted people to know, that it is not a good idea to move the vector tabel to RAM on STM32F0. This took me some time to figure out yesterday.

Now, we should either
* make compilation fail when this feature is enabled on cortex-m0, e.g.
~~~
%% if parameters.vector_table_in_ram and target is cortex_m0
#error "Moving the vector table to RAM is not supported on Cortex-M0 targets"
%% endif
~~~

or

* add support through the same workarounds that mbed is using

ekiwi, 2016-02-09:
However, I would prefer to just fail compilation, as the mbed solution might actually prevent hard faults, when dereferencing nullptr, as they map RAM to address 0, as far as I understand.

In addition to that, they probably require a different linkerfile and I would prefer to avoid that.

ekiwi, 2016-02-09:
@salkinium what solution do you prefer?

salkinium, 2016-02-09:
Yes, I agree with not having a silent failure.
Want to add the `#error` and merge it?

Parameter should be filterable by device properties though, that would be a cleaner solution.
But then again properties are not well implemented, so "clean" is relative ;-P
```xml
<parameter core="cortex-m3|cortex-m4|cortex-m4f|cortex-m7|cortex-m7f" name="vector_table_in_ram" type="bool">false</parameter>
```

commit 2c4a56fd (2016-02-09): [driver] Core: mapping NVIC vectors to RAM is not supported on cortex-m0

This closes #83
commit 25cb72f7 (2016-02-09): [driver] Core: mapping NVIC vectors to RAM is not supported on cortex-m0

This closes #83
closed 2016-02-09 by ekiwi