roboterclubaachen/xpcc #85: Matrix/Vector Multiplications

ekiwi, 2016-02-09:
I failed multiple times to multiply a `xpcc::Matrix4f` with a `xpcc::Vector4f`.

~~~{.cpp}
const xpcc::Vector4f v0;
const xpcc::Matrix4f m0;
const auto v1 = m0 * v0;
~~~

fails with the following error:

~~~
main.cpp: In function ‘int main()’:
main.cpp:46:21: error: ambiguous overload for ‘operator*’ (operand types are ‘const Matrix4f {aka const xpcc::Matrix<float, 4u, 4u>}’ and ‘const Vector4f {aka const xpcc::Vector<float, 4u>}’)
  const auto v1 = m0 * v0;
                     ^
/.../xpcc/src/xpcc/math/geometry/vector4_impl.hpp:826:1: note: candidate: xpcc::Vector<T, 4u> operator*(const xpcc::Matrix<T, 4u, 4u>&, const xpcc::Vector<T, 4u>&) [with T = float; U = float]
 operator * (const xpcc::Matrix<T, 4, 4> &lhs, const xpcc::Vector<U,4> &rhs)
 ^
/.../xpcc/src/xpcc/math/geometry/vector4.hpp:269:28: note: candidate: xpcc::Vector<T, 4u> xpcc::operator*(const xpcc::Matrix<T, 4u, 4u>&, const xpcc::Vector<T, 4u>&) [with T = float; U = float]
  static inline Vector<U,4> operator * (const Matrix<T, 4, 4> &lhs, const Vector<U,4> &rhs)
                            ^
/.../xpcc/src/xpcc/math/geometry/vector4.hpp:263:28: note: candidate: xpcc::Vector<T, 4u> xpcc::operator*(const U&, const xpcc::Vector<T, 4u>&) [with U = xpcc::Matrix<float, 4u, 4u>; T = float]
  static inline Vector<T,4> operator * (const U &lhs, const Vector<T,4> &rhs)
~~~

Using `asTransposedMatrix` which I found in [vector4.hpp](https://github.com/roboterclubaachen/xpcc/blob/9a5213e4aa06819b2e1eeb7e423039b1ecbac0ba/src/xpcc/math/geometry/vector4.hpp#L271):

~~~{.cpp}
const xpcc::Vector4f v0;
const xpcc::Matrix4f m0;
const auto v1 = m0 * v0.asTransposedMatrix();
~~~

fails with the following error:

~~~
main.cpp: In function ‘int main()’:
main.cpp:46:21: error: no match for ‘operator*’ (operand types are ‘const Matrix4f {aka const xpcc::Matrix<float, 4u, 4u>}’ and ‘const xpcc::Matrix<float, 1u, 4u>’)
  const auto v1 = m0 * v0.asTransposedMatrix();
                     ^
/.../xpcc/src/xpcc/math/matrix_impl.hpp:295:1: note: candidate: xpcc::Matrix<T, WIDTH, HEIGHT> xpcc::Matrix<T, ROWS, COLUMNS>::operator*(const T&) const [with T = float; unsigned char ROWS = 4u; unsigned char COLUMNS = 4u]
 xpcc::Matrix<T, ROWS, COLUMNS>::operator * (const T &rhs) const
 ^
/.../xpcc/math/matrix_impl.hpp:295:1: note:   no known conversion for argument 1 from ‘const xpcc::Matrix<float, 1u, 4u>’ to ‘const float&’

/.../xpcc/src/xpcc/math/matrix.hpp:179:3: note: candidate: template<unsigned char RHSCOL> xpcc::Matrix<T, ROWS, ROWS> xpcc::Matrix<T, ROWS, COLUMNS>::operator*(const xpcc::Matrix<T, COLUMNS, RHSCOL>&) const [with unsigned char RHSCOL = RHSCOL; T = float; unsigned char ROWS = 4u; unsigned char COLUMNS = 4u]
   operator * (const Matrix<T, COLUMNS, RHSCOL> &rhs) const;
   ^
/.../xpcc/src/xpcc/math/matrix.hpp:179:3: note:   template argument deduction/substitution failed:
main.cpp:46:45: note:   template argument ‘1u’ does not match ‘#‘integer_cst’ not supported by dump_decl#<declaration error>’
  const auto v1 = m0 * v0.asTransposedMatrix();
                                             ^
main.cpp:46:45: note:   ‘const xpcc::Matrix<float, 1u, 4u>’ is not derived from ‘const xpcc::Matrix<float, 4u, RHSCOL>’

/.../xpcc/src/xpcc/math/matrix_impl.hpp:414:1: note: candidate: template<class T, unsigned char WIDTH, unsigned char HEIGHT> xpcc::Matrix<T, WIDTH, HEIGHT> operator*(int8_t, const xpcc::Matrix<T, WIDTH, HEIGHT>&)
 operator * (int8_t lhs, const xpcc::Matrix<T, WIDTH, HEIGHT> &m)
 ^
/.../xpcc/src/xpcc/math/matrix_impl.hpp:414:1: note:   template argument deduction/substitution failed:
main.cpp:46:45: note:   cannot convert ‘m0’ (type ‘const Matrix4f {aka const xpcc::Matrix<float, 4u, 4u>}’) to type ‘int8_t {aka signed char}’
  const auto v1 = m0 * v0.asTransposedMatrix();
                                             ^
/.../xpcc/src/xpcc/math/matrix_impl.hpp:422:1: note: candidate: template<class T, unsigned char WIDTH, unsigned char HEIGHT> xpcc::Matrix<T, WIDTH, HEIGHT> operator*(float, const xpcc::Matrix<T, WIDTH, HEIGHT>&)
 operator * (float lhs, const xpcc::Matrix<T, WIDTH, HEIGHT> &m)
 ^
/.../xpcc/src/xpcc/math/matrix_impl.hpp:422:1: note:   template argument deduction/substitution failed:
main.cpp:46:45: note:   cannot convert ‘m0’ (type ‘const Matrix4f {aka const xpcc::Matrix<float, 4u, 4u>}’) to type ‘float’
  const auto v1 = m0 * v0.asTransposedMatrix();
                                             ^
/.../xpcc/src/xpcc/math/geometry/vector4_impl.hpp:819:1: note: candidate: template<class T> xpcc::Vector<T, 4u> operator*(const T&, const xpcc::Vector<T, 4u>&)
 operator * (const T &lhs, const xpcc::Vector<T,4> &rhs)
[... 47 lines cut ...]
main.cpp:46:45: note:   ‘const xpcc::Matrix<float, 1u, 4u>’ is not derived from ‘const xpcc::Vector<T, 3u>’
  const auto v1 = m0 * v0.asTransposedMatrix();
                                             ^
/.../xpcc/src/xpcc/math/geometry/vector2_impl.hpp:601:1: note: candidate: template<class U> xpcc::Vector<T, 2u> xpcc::operator*(float, const xpcc::Vector<T, 2u>&)
 xpcc::operator * (float scale, const Vector<U, 2> &vector)
 ^
/.../xpcc/src/xpcc/math/geometry/vector2_impl.hpp:601:1: note:   template argument deduction/substitution failed:
main.cpp:46:45: note:   cannot convert ‘m0’ (type ‘const Matrix4f {aka const xpcc::Matrix<float, 4u, 4u>}’) to type ‘float’
  const auto v1 = m0 * v0.asTransposedMatrix();
                                             ^
/.../xpcc/src/xpcc/math/geometry/vector1.hpp:122:29: note: candidate: template<class U, class T> xpcc::Vector<T, 1u> xpcc::operator*(const U&, const xpcc::Vector<T, 1u>&)
  static inline Vector<T, 1> operator * (const U &lhs, const Vector<T, 1> &rhs)
                             ^
/.../xpcc/src/xpcc/math/geometry/vector1.hpp:122:29: note:   template argument deduction/substitution failed:
main.cpp:46:45: note:   ‘const xpcc::Matrix<float, 1u, 4u>’ is not derived from ‘const xpcc::Vector<T, 1u>’
  const auto v1 = m0 * v0.asTransposedMatrix();
                                             ^
~~~

Now my questions to @salkinium and @dergraaf are:
* Am I using the classes correctly?
* Are these classes intended to be used this way?
* Why is there no unittest for this?

Kevin

salkinium, 2016-02-09:
> - Am I using the classes correctly?

This needs to be ('cos linear algebra):
```cpp
const xpcc::Vector4f v0;
const xpcc::Matrix4f m0;
const auto v1 = m0 * v0.asMatrix();
```

xpcc::Vector could benefit from a conversion operator here.
(or xpcc::Matrix from another `*` operator overload).

> - Are these classes intended to be used this way?

Yes. Although to be honest, we are not the original author of these classes, they [are from here](http://www-etud.iro.umontreal.ca/~petitg/cpp/).
[Looking at parts of the source code](https://github.com/roboterclubaachen/xpcc/blob/develop/src%2Fxpcc%2Fmath%2Fmatrix_impl.hpp#L106-L115) I think the code could benefit a lot from C++11 and constexpr functions.

> - Why is there no unittest for this?

There are unittests, but perhaps they don't cover this use case?
- [xpcc::Matrix](https://github.com/roboterclubaachen/xpcc/tree/develop/src/xpcc/math/test)
- [xpcc::Vector](https://github.com/roboterclubaachen/xpcc/tree/develop/src/xpcc/math/geometry/test)

salkinium, 2016-02-13:
@ekiwi I'm closing this, please reopen if required.

closed 2016-02-13 by salkinium

salkinium, 2016-02-22:
@ekiwi There are other, more up-to-date implementations (using C++11) of Matrix/Vector/Quaternion here: https://github.com/cbiffle/etl/tree/master/math
Might be worth to replace our implementation.