roboterclubaachen/xpcc #95: VL6180 driver: Unaligned memory access on cortex-m0

rleh, 2016-02-16:
in `src/xpcc/driver/position/vl6180.hpp:323` the following code results in a hard fault because of unaligned memory access:
```
uint16_t* rawData = reinterpret_cast<uint16_t*>(data+1);
uint16_t value = xpcc::fromBigEndian(*rawData);
```

A possible fix is replacing the two lines with `uint16_t value = (data[1] << 8) | (data[2]);`

ekiwi, 2016-02-16:
I can confirm this issue and I like the suggested fix.
As data is a `uint8_t` it needs to be accessed as such. On a cortex-m0, casting to pointer of smaller types is ok, casting to pointers of wider values is not ok, as it violates access restrictions.

salkinium, 2016-02-16:
Hm, this is a big issue.
There is quite [a lot of code that uses `reinterpret_cast`](https://github.com/roboterclubaachen/xpcc/search?utf8=✓&q=reinterpret_cast) to access data fields, especially in XPCC (the protocol).
If we fixed this here, wouldn't there be a bunch of hardfaults later because of `reinterpret_cast` usage on packed structs (like here) in XPCC?

I strongly favor the current code, because it's a lot more readable and therefore maintainable.
16bit unaligned endianness conversion is still ok, but with 32bit values this just results in maintenance overhead.

Perhaps a better alternative would be a `xpcc::reinterpret_cast` template function:
On CM0 it does the access byte-wise, but on CM3 just defaults to `reinterpret_cast`.
This would make this issue go away in a transparent way and all of the xpcc and robot code can be updated with one find and replace.

I cannot find a similar solution with 5mins of googling, so double-check me please: Is that an insane solution? I think it's doable, but maybe I made a reasoning mistake?

ekiwi, 2016-02-16:
> I strongly favor the current code, because it's a lot more readable and therefore maintainable.

Actually, I disagree:

~~~{.cpp}
uint16_t value = (data[1] << 8) | (data[2])
~~~

is much simpler and much more readable than:

~~~{.cpp}
uint16_t* rawData = reinterpret_cast<uint16_t*>(data+1);
uint16_t value = xpcc::fromBigEndian(*rawData);
~~~

`reinterpret_cast` should only be used in very rare cases, if it is really necessary. It is basically telling the compiler: I know better then you!

And please remember: An unaligned access will work on Cortex-M3/M4, but might still result in two bus transactions and thus a performance hit, so I see no reason for the original code.

ekiwi, 2016-02-16:
> Perhaps a better alternative would be a xpcc::reinterpret_cast template function:
> On CM0 it does the access byte-wise, but on CM3 just defaults to reinterpret_cast.
> This would make this issue go away in a transparent way and all of the xpcc and robot code can be updated with one find and replace.

I do not think, that this will work, as the access is not performed in the first, but in the second line. As the pointer type is declared to be `uint16_t*` we cannot really transparently change that.
We should rather check all occurrence of `reinterpret_cast<>`  on whether they are strictly necessary or just use of C idoms or premature optimizations.

salkinium, 2016-02-17:
> is much simpler and much more readable than:

Excuse me?!

You are manually building a 16bit variable out of a 8bit array **and** doing endianness conversion at the same time.
That is **not** simple and definitely **not** readable. I think "confusing" is a better word.

I have to look up endianness order every single time, they are just really difficult to remember.
Besides, the `xpcc::fromBigEndian` is not the problem here, the unaligned access is. Let's separate the concerns here.

The code could have also been:
```cpp
uint16_t rawData = *reinterpret_cast<uint16_t*>(data+1);
uint16_t value = xpcc::fromBigEndian(rawData);
```
and we would probably not be talking about endianness conversion.

> And please remember: An unaligned access will work on Cortex-M3/M4, but might still result in two bus transactions and thus a performance hit, so I see no reason for the original code.

You want to nitpick about the AHB-Lite bus performance? With me? Ok. *takes off gloves, opens ARMv7-M Architecture Reference Manual v3 to Chapter 2.2.5 and AMBA 3 AHB-Lite Protocol Specification v1.0 to Chapter 3.5, clears throat*
1. Unaligned access is done only when required. So it is done in 1 access if possible, your solution always requires two bus accesses (unless GCC recognizes it as an idiom and optimizes it away on !CM0. Might be interesting to check that actually ;-).
2. Unaligned access is done in a 16-beat incrementing burst transfer. This means it cannot be interrupted  by other bus masters (not an issue on AHB-Lite, there is only one Master) and for the second transfer the address phase is not required, because it's an incrementing transfer.

:smiling_imp: 
None of this really matters on a **48MHz** device. The point is that I shouldn't have to deal with alignment issues manually.

> As the pointer type is declared to be uint16_t* we cannot really transparently change that.

Yes. We. Can.

So you'd need a `uint_unaligned16_t` type with an overloaded dereferencing operator that does the access correctly (same for 32bit, etc). You'd have the chance to fix **writing** to unaligned locations as well, btw. This isn't magic.
This would be the best solution, since it is transparent to the user, it can be unit tested (!!!), it's incurs no overhead on CM3, it's native C++, it's easily portable to AVR and other crazy platforms, and it's a drop-in replacement for what we have now.

> We should rather check all occurrence of reinterpret_cast<> on whether they are strictly necessary or just use of C idoms or premature optimizations.

This isn't about optimization, this is about correctness and maintenance.
I personally couldn't care less about the performance of unaligned access in a frikkin' **48MHz** device, but I **do** care about usable and correct endianness conversion.

Convinced? Or is moar lobbying required? ;-)

ekiwi, 2016-02-17:
> I have to look up endianness order every single time, they are just really difficult to remember.

The code that I presented above actually is not dependent on the endianess of the target architecture. In addition to that it follows a common idom in embedded programming, which makes it easier to comprehend then some pointer magic.

> So you'd need a uint_unaligned16_t type with an overloaded dereferencing operator that does the access correctly (same for 32bit, etc).

Yes, this would work, however, we would have to change more than just the `reinterpret_cast`.

> This isn't about optimization, this is about correctness and maintenance.

Yes, here we both agree, I actually brought up the performance point, because I thought, that some of the `reinterpret_cast` occurrences in our code base represent premature optimizations.

ekiwi, 2016-02-17:
I had a lot of time to think about alignment issues and how they can be avoided during my Bachelor thesis, and actually it is not too complicated:

### gcc alignment
* for structs, as long as `packed` isn't used, gcc generates correctly aligned code
* uint32_t, uint16_t and uint8_t arrays are always correctly aligned for the underlying data type, if they are declared in static memory
* as far as I know, the dynamic memory allocator distributes word aligned memory

### operations that should always work
* accessing `uint16_t[14]` as `reinterpret_cast<uint8_t*>`
* accessing `uint32_t[14]` as `reinterpret_cast<uint16_t*>`
* accessing `uint32_t[14]` as `reinterpret_cast<uint8_t*>`
* casting `void*` to `reinterpret_cast<uint32_t>` (better would be `usize_t`)

### operations that (in my understanding of the memory modell) are not guaranteed to succeed
* accessing `uint8_t[14]` as `reinterpret_cast<uint16_t*>`
* accessing `uint16_t[14]` as `reinterpret_cast<uint32_t*>`

### summary
Thus as long as you do not `reinterpret_cast` to pointers with wider types everything works fine. A lot of casts in the xpcc code base actually do not violate that rule, thus they should be fine.

ekiwi, 2016-02-17:
To come back to our little example: we have a byte buffer that is filled with single bytes received via I2C, which is why the datatype of `data` makes sense. Now you want to read a 16bit wide unsigned Integer from the byte data. The format of which is specified in the manufacturers data sheet.

For this type of situation, I do not think, that `reinterpret_cast` is the correct idom to use in our code. I would rather like to have a accessor method, that takes a byte buffer and can read the manufacturers format. Something like:

~~~{.cpp}
auto value = data.read<uint16_t>(1);
~~~

to archive this, we can use a template class (the naming and correct implementation still needs a lot of improvement):

~~~{.cpp}
// private member of vl6180
BigEndianByteBuffer<3> data;
~~~

salkinium, 2016-02-17:
> The code that I presented above actually is not dependent on the endianess of the target architecture.

Yes it is, it's just hidden in the compiler…

> In addition to that it follows a common idom in embedded programming, which makes it easier to comprehend then some pointer magic.

"Nothing is so perfect that it can't be complained about".
Also, since when does xpcc follow "common idoms in embedded programming"?
Did I miss a memo?
:stuck_out_tongue_winking_eye: 

> Yes, this would work, however, we would have to change more than just the `reinterpret_cast`.

Huh? This should all work.
```cpp
uint16_t rawData1 = *reinterpret_cast<uint_unaligned16_t*>(data+1);
uint_unaligned16_t *rawData2 = reinterpret_cast<uint_unaligned16_t*>(data+1);
uint16_t rawData3 = *rawData2;
```

Anyways. Ok, fine, forget my beautifully over-engineered `uint_unaligned16_t` solution :sob:

How about we add a `uint16_t xpcc::fromBigEndian16(uint8_t *data)` function that abstracts this?
That would mean no (visible ;-) pointer magic and it will work correctly for CM0 and can be optimzed for !CM0.

The only annoying thing is that you'd have to cast the return type manually to signed, because overloading doesn't work for return types.

salkinium, 2016-02-17:
> I had a lot of time to think about alignment issues and how they can be avoided during my Bachelor thesis, and actually it is not too complicated:

I remember you cursing about it. 
Lets use that knowledge and abstract it in one place, so that we never have to worry about it ever again.

> `auto value = data.read<uint16_t>(1);`

Sure, looks great!

But why is `BigEndianByteBuffer<3>.read<uint16_t>(1)` so specialized?
Why wouldn't it just be a ~~`std::`~~ `xpcc::array<Type, Length>` and the `Type` would decide how it is accessed?
If we take the modularity approach further, wouldn't it result in something like: `xpcc::array< xpcc::big_endian_t< uint_ualign16_t >, 3 >`?
(ZOMG, We are approaching `SimpleBeanFactoryAwareAspectInstanceFactory` length declarations1!11)

I just don't see why the array is special and not the type.
Specializing the type is just cleaner, I can then write:
```cpp
struct __xpcc_packed {
   uint8_t foo;
   xpcc::big_endian_t< uint_ualign16_t > bar;
}
```
and expect it to work, right?
Does that look terrible?

dergraaf, 2016-02-17:
> ~~~{.cpp}
> auto value = data.read<uint16_t>(1);
> ~~~

Actually that is somewhat what I implemented at work to solve this exact
problem. Not with an index but with an streaming interface. It works well,
although there is no easy way to specify the structure of the data you're
reading. It is only defined through the order in which you call the read
functions.

salkinium, 2016-02-18:
Well, I'm still in favor of `xpcc::big_endian_t< uint_ualign16_t >` because it's modular and ties the implementation to the type and not to a function, which should then be easier to transparently use.

However, since this is a blocker for using this driver (and probably a few others) on the Cortex-M0, and that I won't have the time to fix this during the next weeks, I suggest we use:
```
// manually building uint16_t to avoid unaligned access problems on CM0
uint16_t value = (data[2] << 8) | (data[1]);
value = xpcc::fromBigEndian(value);
```
for now.

I explicitly am not in favour of doing only `uint16_t value = (data[1] << 8) | (data[2]);`, because it conflates two unrelated problems: unaligned access and endianness conversion.

salkinium, 2016-02-28:
Isn't it in this case easier to let the compiler handle unaligned access?

Adding a union on the raw `data[3]` like so:
```cpp
union {
	uint8_t data[3];
	struct {
		uint8_t distance;
		uint16_t raw_lux;
	};
};
```
and then accessing `raw_lux` should just work, no?
```cpp
uint16_t value = xpcc::fromBigEndian(raw_lux);
```

ekiwi, 2016-02-28:
> Isn't it in this case easier to let the compiler handle unaligned access?

1.) I do not think, that a 10 line solution that involves a `reinterpret_cast` and the word `endian` can be called easier than a 1 line solution that follows a well established pattern and that should work on every standard compliant C++ compiler.

2.) Your solution will probably not work in all cases. According to the [C++ Standard Working Draft](http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2014/n4296.pdf)  (ISO N4296, section 9.2.13):

> Nonstatic data members of a (non-union) class with the same access control (Clause 11) are
> allocated so that later members have higher addresses within a class object. The order of
> allocation of non-static data members with different access control is unspecified (Clause 11).
> Implementation alignment requirements might cause two adjacent members not to be allocated
> immediately after each other;

Thus, on a `cortex-m0` the compiler might leave one byte of space between `distance` and `raw_lux` in order to properly align `raw_lux`. You can probably use the gcc extension `packed` to make this code work.

However, I have to confess, that I do not see the point of this, when there is a very portable and in my opinion easy to understand solution with well defined semantics.
While your solution requires knowledge about the memory model and endianness, the solution that I suggested, just requires you to understand, that there are two bytes that, out of one reason or another, are in separate locations. However, they have a combined meaning, one of them has to be multiplied by 255 and then added to the other, to form a valid 16bit integer.

salkinium, 2016-02-28:
> However, I have to confess, that I do not see the point of this, when there is a very portable and in my opinion easy to understand solution with well defined semantics.

Ok, since we need a fix, so let's fix it this way.

strongly-typed, 2016-02-29:
Another random idea:

As already nearly 99% of xpcc's code is generated before compile time:
What about dealing with alignment and endianness before compiling?

salkinium, 2016-02-29:
> As already nearly 99% of xpcc's code is generated before compile time:
> What about dealing with alignment and endianness before compiling?

Not sure if trolling or…

> one of them has to be multiplied by 255 and then added to the other

(1 << 8) = 256 btw.

Anyway, @ekiwi will fix it this way for now, since this is a blocker.
But particularly for 32bit this can become error prone quickly:
```cpp
// big endian memory or little endian memory format
uint32_t u32 = (uint32_t(data[0]) << 24) | (uint32_t(data[1]) << 16) | (uint32_t(data[2]) << 8) | data[3];
// too lazy to look this up now.
uint32_t u32 = (uint32_t(data[3]) << 24) | (uint32_t(data[2]) << 16) | (uint32_t(data[1]) << 8) | data[0];
```

salkinium, 2016-02-29:
@strongly-typed: I think you are referring to the generated drivers for external ICs.
That is a concept where one describes the memory maps and transport layer (and perhaps even functionality) in an abstract format and then generates the C++ representations and code for that.
Would be nice, since that would make it possible to un-specialize the drivers from xpcc and generate drivers of similar quality for other systems (ie. linux, arduino, mbed, FPGA?).

But that is not on my list of priorities right now, and doesn't help in this case.

salkinium, 2016-03-08:
Using `xpcc::asUnaligned` would make this look like this:
```cpp
uint16_t rawData = *xpcc::asUnaligned<uint16_t>(data+1);
uint16_t value = xpcc::fromBigEndian(rawData);
```

This would also work with `using uint_unaligned16_t = xpcc::unaligned_t<uint16_t>;`:
> ```cpp
> uint16_t rawData1 = *reinterpret_cast<uint_unaligned16_t*>(data+1);
> uint_unaligned16_t *rawData2 = reinterpret_cast<uint_unaligned16_t*>(data+1);
> uint16_t rawData3 = *rawData2;
> ```

Note that these two have subtly different semantics but do the same thing:
```cpp
auto value = *xpcc::asUnaligned<uint16_t>(data+1);
auto value = data.read<uint16_t>(1);
```

closed 2016-03-13 by salkinium

ekiwi, 2016-03-28:
@salkinium a small followup on unaligned access: http://kristerw.blogspot.de/2016/03/pointer-casts-in-c.html is a very interesting read.

PR #108 (2016-03-13): Add unaligned accessor helper
This adds a `xpcc::unaligned_t` wrapper class, that manages access to the underlying type using byte-wise access.
For access to bytes streams use the `xpcc::asUnaligned<T>` helper class.
See example for usage.

This also fixed alignment issues in `xpcc::SmartPointer`, which should make the XPCC protocol work on CM0 as well.

Fixes #95.

cc @ekiwi @dergraaf @rleh @strongly-typed