roboterclubaachen/xpcc #112: Hard fault handler does not compile without optimization

rleh, 2016-03-14:
Disabling compiler optimization (`"-Os"` -> `"-O0"` in `scons/site_tools/arm.py:117`) produces the following error:
```bash
$ cd examples/stm32f4_discovery/hard_fault
$ scons
/* ... */
Compiling C++: /home/user/xpcc/build/stm32f4_discovery/hard_fault/libxpcc/build/stm32f4_discovery/hard_fault/libxpcc/generated_platform/driver/core/cortex/hard_fault_handler.o
In file included from /home/user/xpcc/src/xpcc/debug/logger.hpp:31:0,
                 from /home/user/xpcc/build/stm32f4_discovery/hard_fault/libxpcc/generated_platform/driver/core/cortex/hard_fault_handler.cpp:37:
/home/user/xpcc/src/xpcc/debug/logger/logger.hpp: In function 'void _hardFaultHandler(const uint32_t*, const uint32_t*)':
/home/user/xpcc/src/xpcc/debug/logger/logger.hpp:95:6: error: local frame unavailable (naked function?)
      *(xpcc::IOStream *) this << msg;
      ^
/home/user/xpcc/src/xpcc/debug/logger/logger.hpp:95: confused by earlier errors, bailing out
scons: *** [/home/user/xpcc/build/stm32f4_discovery/hard_fault/libxpcc/build/stm32f4_discovery/hard_fault/libxpcc/generated_platform/driver/core/cortex/hard_fault_handler.o] Error 1
scons: building terminated because of errors.
```

salkinium, 2016-03-14:
Hm, I've been misusing [`__attribute__((naked))`](https://gcc.gnu.org/onlinedocs/gcc/ARM-Function-Attributes.html#ARM-Function-Attributes):

> This attribute allows the compiler to construct the requisite function declaration, while allowing the body of the function to be assembly code. The specified function will not have prologue/epilogue sequences generated by the compiler. Only basic asm statements can safely be included in naked functions (see Basic Asm). While using extended asm or a mixture of basic asm and C code may appear to work, they cannot be depended upon to work reliably and are not supported. 

It's probably only sheer luck, that the startup script, which is also attributed `((naked))` compiles with `-O0`.
Let me fix that.

salkinium, 2016-03-14:
Please note that for `-O0` the stack size for the hard fault handler stack is probably too small.
It is currently set at 192 bytes, which has been measured with `-Os` only.

salkinium, 2016-03-14:
So the worst stack usage I've seen with `-O0` and without using `__attribute__((naked))` is over 550 bytes, when using the logger.
This is not acceptable and will require a bigger fix than just removing the `naked` attribute.

I'll push a hotfix to remove the naked attribute, so for now don't enable serial logging in the hard fault handler.

salkinium, 2016-03-16:
I haven't found a satisfying answer to this problem yet.

The canonical way of dealing with hardfaults would be to create a GDB corefile for offline debugging later, but [`arm-none-eabi-gdb` does not support corefile creation or consumption](https://sourceware.org/bugzilla/show_bug.cgi?id=14383).
I've raised attention to this issue internally with ARM, let's see what happens.

In the meantime I'm considering using the [CrashCatcher](https://github.com/adamgreen/CrashCatcher), which would enable debugging after the crash.
@daniel-k Do you think the debug server has a problem with a few kB of data like that?

daniel-k, 2016-03-16:
@salkinium 
CrashCatcher looks like an interesting workaround! Some time ago I also investigated on creating corefiles for ARM with the same result.

I suppose a dump of several kB of data on the serial port is no problem at all. Enable setting a prefix and I can dump it directly into a file ;-)

salkinium, 2016-03-22:
The changes to the hard fault handler have been merged.

closed 2016-03-22 by salkinium