roboterclubaachen/xpcc #114: STM32: CAN: add timeout to initialize method

ekiwi, 2016-03-19:
Currently the `initialize` method busy waits for the CAN IP to enter normal mode. If no traceiver is connected or maybe even if there is a bus fault, this wait will block forever.
We should add a timeout and return an error, if the IP did not enter normal mode.

~~~
	// Request leave initialization
	CAN1->MCR &= ~(uint32_t)CAN_MCR_INRQ;
	while ((CAN1->MSR & CAN_MSR_INAK) == CAN_MSR_INAK) {
		// wait for the normal mode
	}
~~~

salkinium, 2016-03-20:
Something like this?
```cpp
int deadlockPreventer = 100000; // kinda depends on CPU frequency
while (((CAN1->MSR & CAN_MSR_INAK) == CAN_MSR_INAK) and (deadlockPreventer-- > 0))
	;
```
Alternatively, if a more accurate timing is required perhaps this?
```cpp
int deadlockPreventer = 1000; // max ~1ms
while (((CAN1->MSR & CAN_MSR_INAK) == CAN_MSR_INAK) and (deadlockPreventer-- > 0))
	xpcc::delayMicroseconds(1);
```

ekiwi, 2016-03-20:
> Alternatively, if a more accurate timing is required perhaps this?

Yes. Something like this.

What we need to discuss is how to adjust our CAN driver interface.
Should the `initialize` [function](https://github.com/roboterclubaachen/xpcc/blob/7b32f74cc43c7bfaf815287902ef2bd4a6baa84d/src/xpcc/architecture/interface/can.hpp#L90) just return `false` if it fails?

Or should we just return the `BusState` from the `initialize` function?
If it is not `BusState::Connected` something went wrong.

We probably should find out what [Can::getBusState](https://github.com/roboterclubaachen/xpcc/blob/7b32f74cc43c7bfaf815287902ef2bd4a6baa84d/src/xpcc/architecture/platform/driver/can/stm32/can.cpp.in#L442) returns after the initialization fails, because no transceiver is connected. If this information is useful it might be worthwhile returning that instead of just a boolean value.

What do you think @salkinium ?

salkinium, 2016-03-20:
> return the BusState from the initialize function?

:+1: As long as the interface an all implementations are updated, this is a good solution.
(I found [STM32](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/can/stm32/can.hpp.in#L104), [LPC11](https://github.com/roboterclubaachen/xpcc/blob/7b32f74cc43c7bfaf815287902ef2bd4a6baa84d/src/xpcc/architecture/platform/driver/can/lpc/c_can.hpp#L54) and [MCP2515](https://github.com/roboterclubaachen/xpcc/blob/7b32f74cc43c7bfaf815287902ef2bd4a6baa84d/src/xpcc/driver/can/mcp2515.hpp#L141)).

Might also be worth removing the `interruptPriority` as it's not part of the interface and just simply confusing. At the very least it should be defaulted to something sensible and not forced into the users face, who probably cannot make a useful decision on this anyways.

salkinium, 2016-03-20:
> Might also be worth removing the interruptPriority as it's not part of the interface

This is also an issue with the [STM32 UART `initialize`](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/uart/stm32/uart.hpp.in#L59) function.
There should be an interface to the NVIC with [methods like these](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/uart/stm32/uart_hal.hpp.in#L158-L176) and all peripherals can inherit and implement these methods.

strongly-typed, 2016-03-20:
Please, don't mess up with another part that is so essential for RCA. There
was already a discussion about this at RCA, and the conclusion was: Waiting
forever is "alternativlos" for RCA here.

When being used the CAN is a somewhat essential component of the robot
(yes, real robots need a working CAN). If there is a problem with the CAN
bus, nothing should happen so that the operator of the robot notices that
he or she must take care about something. Failing silently is a no-go. And
there are no practical error reporting strategies in place, yet.

If you touch this behaviour make sure there are no side effect for the
actual robot running in a competition.

salkinium, 2016-03-20:
> Failing silently is a no-go. And there are no practical error reporting strategies in place, yet.

But isn't the current code failing silently too?
I think its very risky to depend on an xpcc internal implementation detail for something that important.
Adding a return type will allow busy waiting forever outside of the `initialize()` method, or a call to an error function, which would make this non-silently failing.

This is an example for the need of exceptions, since you can ignore a return value, but not an exception. ~~Java anyone?~~ ;-)

> Please, don't mess up with another part that is so essential for RCA.

Well, the `ClockControl` driver [uses timeouts](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/clock/stm32/clock.cpp.in#L60-L64).
If the clock setup fails when using `SystemClock::enable()`, the previous clock will remain active and an error is returned.

How do we deal with such fatal failures?

ekiwi, 2016-03-20:
> But isn't the current code failing silently too?

That is exactly, what @daniel-k and I thought as well. We agreed, that it would be better to display an error via UART that is visible in the log of our Debug Server than to just freeze the robot.
Maybe we can, in addition to that, implement some form of error reporting to the display.

strongly-typed, 2016-03-21:
> But isn't the current code failing silently too?

Silently in the sense of "no debug output": yes, you are absolutely right. This might not be optimal. Adding some debug output by UART is something different from changing the infinite loop to a finite loop. +1 for adding debug output. -1 for changing the loop without other measures, see below.

Silently in the sense of "User may miss that some part of the Minimum Equipment List (MEL) is not working": Strong No. It does not make sense to reposition a robot without a working CAN bus.

> implement some form of error reporting to the display.

By sending a message from the main board to the display over the CAN bus ? Hmm, ...

> How do we deal with such fatal failures?

It should be avoided under all circumstances that if a piece of the MEL is not working and the robot seems to behave normally, e.g. doing its repositioning movement.

Remember the situation when it was necessary to check manually if a sensor connected by I2C by triggering this sensor and observing the LED? Should the user test the CAN bus in every tournament before starting the robot?

> How do we deal with such fatal failures?

Blocking is alright during initialisation. That signalises to the user undoubtedly that there is a serious problem that must be dealt with immediately.

> will allow busy waiting forever outside of the initialize()

Yes, but this change must go hand in hand with the proposed change to xpcc. And error reporting, error handling and safe recovery is one of the key strengths of RCA's robot code. `</irony>`

> If the clock setup fails when

This is another item from the MEL, but less likely to fail than a CAN bus with a differential impedance of 10 Ohms or less ...

> I think it's very risky 

I think it's very risky to change the behaviour of code without considering all side effects this change may have. If you make sure all components of the MEL are up and running before announcing "Ready to start" on the robot, it's all fine, safe and sound.

salkinium, 2016-03-21:
Hang on, I need some popcorn for the tone of this thread.

salkinium, 2016-03-22:
>> Please, don't mess up with another part that is so essential for RCA.

> And error reporting, error handling and safe recovery is one of the key strengths of RCA's robot code.

Considering the limited human resources we have available and the challenges that needed solving, we can be very proud of how stable, well tested (#101) and *safe* both xpcc and our robot code is. Even more so considering that we are using cooperative multitasking for all of it and it is still reasonably easy to wrap your head around how it works, despite the complexity of the problems and solutions.

And now back to your regularly scheduled program…

## The curious case of xpcc and the missing error model

In hindsight it is quite apparent that xpcc and therefore also the robot code is missing a good error model.

xpcc's current error model is returning error codes and using `static_assert`. We have never really considered runtime assertion, nor dealing with hardware exceptions like hardfault handlers (#113).
There were some efforts made to make I2C failures recoverable, but its state machine approach cannot be made to fit protothreads.
Returning error codes clearly shifts the burden of error checking to the programmer and this is not a good design.

#### Research

[Joe Duffy wrote a fantastically detailed article on the many considerations that go into a error model.](http://joeduffyblog.com/2016/02/07/the-error-model/)
Of course we can and will not roll out our own language with a custom compiler so we are stuck with C++ and C++ exceptions are clearly not the answer.

However, there is are couple of things in there that resonates very strongly with me:

1. "Reliability, Fault-Tolerance, and Isolation": If you look at how we use XPCC (the protocol) in our robots, failure of actions is a very prominent consideration and we strongly encourage dealing with it. The pain to deal with failure has gotten a lot less with the Activities concept and the accompaning use of the Resumable Postman. The fact that we have gotten to this point is in itself already quite an amazing thing.
1. "Bugs Aren’t Recoverable Errors!": This, I think, is the single most important take away from this post. When we are talking about the SystemClock or the CAN bus not initializing and timing out, these are bugs and you cannot recover from them. I2C bus errors on the other hand are fully recoverable (in theory at least, there is that one corner case ;-) and they can happen in normal operation.
1. "Abandonment": xpcc doesn't have a concept of abandonment. We do not even have runtime assertions. There are a couple of `while(1)` loops in the vector table (and hard fault handler), but there is no controlled teardown (with reporting) of failures.

#### Proposal

I propose a two phase process to prototype and enable runtime assertions with abandonment in xpcc.

1. Add `xpcc_assert_debug` and `xpcc_assert` macros, that check a boolean condition, and on failure report this to `XPCC_LOG_ERROR` perhaps in a similar way that the unit test assertions report before dying or rebooting.
1. Add the ability to attach custom handlers to deal with assertions and allow to ignore assertion failures if required.

`xpcc_assert` will always be an active assertion.
`xpcc_assert_debug` will only be active in a Debug build. I'm open to whether this should be a global setting, like in most build environments, or whether this should be decided on a per-file basis, as we are doing with `XPCC_LOG_LEVEL`.

Assertions should be given an error identifier as a `char[]`. It is a lot simpler to encode structured information using stringly-typed programming than using binary structures. So for example, a failure in the SystemClock could have the string `"clock.pll"` and you know exactly where the failure occurred. Similarly, `"can.init"` gives you the scope (`"can"`) and type of failure (`"init"`) in just a few bytes.
Note that this can consume significantly less memory than using a stringified version of the test condition as or a "pretty" function string, or even just line and file.

But most importantly, it allows custom failure handlers to assess the scope and type of failure programmatically.
By default all failing assertions would result in abandonment and the program would die.
My experience with designing ARMmbed/utest suggests that sometimes you need to have a custom failure handler.

An assertion failure handler could have the signature `abandonment_t(char[] id)` and it would assess the failure by parsing the id string and return what type of abandonment it would perfer.
For example, a custom failure handler can compare the first three characters and if they are equal to `"can"`, it will sound a really annoying buzzer.
However, if an assertion fails during a game that is not super critical, for example we lost a distance sensor and some values are completely out-of-range, then you want to report that failure, but ignore it.
A failure handler can return `Fail`, `DontCare` or `Ignore` depending on whether the error identifier means something to it.

There could be multiple failure handlers, which would then be called one by one in by the assertion.
If any or them `Fail` or all of them `DontCare`, then the process is abandoned. Otherwise if one or more of them `Ignore` the assertion, execution continues.

The way to implement multiple failure handlers would be to add a section to the linkerscript `.assert`, which contains the pointers to all failure handlers.
[... cut ...]
	while (((CAN1->MSR & CAN_MSR_INAK) == CAN_MSR_INAK) and (deadlockPreventer-- > 0))
    	xpcc::delayMicroseconds(1);
    
    xpcc_assert(deadlockPreventer, "can.init");
    return deadlock_preventer;
}

```

Somewhere else a custom failure handler is declared and registered:

```cpp
Abandonment can_assertion_handler(char[] name)
{
	if (strncmp("can", name, 3) == 0) {
		CanFailBuzzer::enable();
		return Abandonment::Fail;
	}
	
	return Abandonment::DontCare;
}
ASSERTION_HANDLER(can_assertion_handler);
```

salkinium, 2016-03-23:
As additional food for thought, here are the causes of abandonment in Midori and the possible implementation in xpcc:

- An incorrect cast. ~> undetected
- An attempt to dereference a `null` pointer. ~> Hard Fault
- An attempt to access an array outside of its bounds. ~> assertion
- Divide-by-zero. ~> Hard Fault
- An unintended mathematical over/underflow. ~> assertion with saturated math classes?
- Out-of-memory. ~> assertion in allocator
- Stack overflow. ~> Hard Fault
- Explicit abandonment. ~> assertion, duh
- Contract failures. ~> n/a, undetected
- Assertion failures. ~> assertion

ekiwi, 2016-03-23:
@salkinium I appreciate your idea, but could we please move the discussion about error handling in `xpcc` to the mailing list, or at least a separate issue?

salkinium, 2016-03-23:
Sure.

For now just add the return value to `initialize` and adapt the robot code.

salkinium, 2016-05-22:
Any intention to implement this? @ekiwi

ekiwi, 2016-05-22:
@salkinium  to be honest, not next week or the week after that

PR #220 (2017-03-12): [stm32] Assert timeout in CAN initialization.
Fixes #114.

FYI: cc @ekiwi @strongly-typed @daniel-k
closed 2017-03-12 by salkinium