roboterclubaachen/xpcc #116: Hardfault handler when debugger connected

daniel-k, 2016-04-05:
So the new hardfault handler can tell if a debugger is connected and therefore decide whether to output some logging info or not. This is nice in theory, but since our debug-server is connected all the time to our robot, it never prints any logging info since a debugger is always detected.

NB: this especially happens if the processor is running without an active debug session, e.g. after a power cycle.

salkinium, 2016-04-05:
See https://github.com/roboterclubaachen/xpcc/commit/0d06f84fb2c7c7a002795334acaac2700d1cedd2.

The debug server runs its own openocd script? If yes, this might need to be adapted like that.

salkinium, 2016-04-05:
Also see [the detailed description of this feature here](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/hard_fault.sx.in#L49-L82).

salkinium, 2016-04-07:
@daniel-k did that solve the issue?

daniel-k, 2016-04-09:
Sorry for the inactiveness. Didn't have the time yet to look into this again.

salkinium, 2016-05-05:
I'm closing this, please reopen, if the issue is not resolved by a change of the openocd script.

closed 2016-05-05 by salkinium