roboterclubaachen/xpcc #118: add -ftabstop=4 as compiler flag

ekiwi, 2016-04-28:
As this is the standard for our code base, informing the compiler might improve the accuracy of the gcc6 misleading indent warnings.

salkinium, 2016-04-28:
Or just replace all tabs with spaces. 
🔥 🔥 🔥 

🚒 🚒 🚒

salkinium, 2016-05-12:
Is this a GCC 6 only flag?

ekiwi, 2016-05-12:
> Is this a GCC 6 only flag?

might be possible ... gcc has a history of introducing flags just when they are needed instead of ignoring them in older versions, see our history with `-fdiagnostics-color=auto` which killed xpcc for gcc 4.8.

ekiwi, 2016-09-03:
Let's not add this flag for the moment, as I do not feel like detecting the `gcc` version in order to only provide this flag for the versions which support it.
Let's reopen this issue, once we are actually getting misleading indent warnings.

closed 2016-09-03 by ekiwi