roboterclubaachen/xpcc #122: TLSF allocator makes xpcc quite big

strongly-typed, 2016-05-07:
Blinky is bloated from 2044 bytes to 32920 bytes flash usage.
This ist only 3% on STM32F407, but about 50% of flash on STM32F103.

	➜  blink git:(c17840c) scons
	[...]
	Size after:
	Creating load file for Flash: /Users/usr/tmp/xpcc/build/stm32f4_discovery/blink/blink.hex
	Memory Usage
	------------
	Device: stm32f407vg
	
	Program:    2044 bytes (0.2% used)
	(.data + .fastdata + .reset + .rodata + .text)
	
	Data:       3096 bytes (1.6% used) = 24 bytes static (0.0%) + 3072 bytes stack (1.6%)
	(.bss + .data + .fastdata + .noinit + .stack + .vectors)
	
	Heap:     131060 bytes (66.7% available)
	(.heap1 + .heap2)
	
	scons: done building targets.
	
After adding TLSF in commit ffa4e1b
	
	➜  blink git:(ffa4e1b) scons
	[...]
	Size after:
	Creating load file for Flash: /Users/usr/tmp/xpcc/build/stm32f4_discovery/blink/blink.hex
	Memory Usage
	------------
	Device: stm32f407vg
	
	Program:   32920 bytes (3.1% used)
	(.data + .fastdata + .reset + .rodata + .text)
	
	Data:       5360 bytes (2.7% used) = 2288 bytes static (1.2%) + 3072 bytes stack (1.6%)
	(.bss + .data + .fastdata + .noinit + .stack + .vectors)
	
	Heap:     191248 bytes (97.3% available)
	(.heap0 + .heap1 + .heap2)
	
	scons: done building targets.

Probably the `printf` bloats the thing as `_vfprintf_r` is included.

	➜  blink git:(ffa4e1b) scons symbols
	scons: Reading SConscript files ...
	scons: done reading SConscript files.
	scons: Building targets ...
	Show symbols for '/Users/usr/tmp/xpcc/build/stm32f4_discovery/blink/blink.elf':
	[...]
	134224528 00005764 T _vfprintf_r
	scons: done building targets.

@salkinium Could you please remove the `printf` debugging output or make it at least configurable?

ekiwi, 2016-05-07:
For targets that are low on memory, the block allocator used to be default as far as I know as we were afraid to use the one included with newlib.

salkinium, 2016-05-07:
😨😱

strongly-typed, 2016-05-07:
`examples/stm32f4_discovery/blink` is growing constantly in flash size.

[image]

I acknowledge that there was some very smart functionalities added (hardfault handler, safe stack, ...). But I think I would not hurt to look at flash wasters. 

I can remember a tool in "old xpcc" that did that kind of profiling and highlighted flash and RAM usage changes from release to release.

The graph is just a quick hack:

	#!/usr/bin/env/bash

	export SCONSFLAGS=-j4
	# git checkout develop

	n=1

	while [ $n -le 1000 ]
	do
		size=`scons | grep "Program" | awk '{ print $2 }'`
		if [ -e $size ]; then size="NaN"; fi

		hash=`git rev-parse HEAD`
		log=`git log --oneline -1`

		# Log
		echo ${n}"," ${size}"," ${hash}"," \"${log}\" >> flashsize.csv

		# Advance / retreat
		git checkout HEAD~1

		let n=n+1
	done

	# Undo and cleanup
	git checkout develop

Then plotted with gnuplot

    set yrange [1000:2500]
    set xrange [350:1] reverse
    set key left
    set xlabel "Commits before commit ffa4e1b"
    set ylabel "Flash size in bytes"
    plot "flashsize.csv" with steps lw 2

I was not able to compile any commit before 9ae68b8 with my current gcc on OSX. Could be interesting to track flash and RAM usage over commits and different compilers.

salkinium, 2016-05-08:
@strongly-typed could you dump the raw `flashsize.csv` into a https://gist.github.com and link it here?

salkinium, 2016-05-08:
Neva mind, here it is: https://gist.github.com/salkinium/e5a53f7bcc9984cfa1698f5e79df0a7c

ekiwi, 2016-05-08:
Has anyone tried enabling link time optimizations for `xpcc`?
I just discovered, that setting the `-flto` flag is not enough, as this leads to some functions not being found.

Some of the increase in code size could be due to @salkinium including some "board support package" functions that might be included with the resulting binary.

salkinium, 2016-05-08:
I'm working on an annotated graph for flash size and static RAM size for some clarity.

Just to be clear, I don't think 2.4kB of Flash usage with 28B of static RAM usage (!) + 3kB of stack (which is completely oversized btw) is a real issue.
The increase is flash size is well worth the additions and **fixes** of features that are critical to a stable ARM Cortex-M port.

There is also a massive tradeoff in development cost vs. Flash size:
Think about Flash Latency, which added 208B of Flash. It could have been dramatically reduced by pushing the lookup table into compile time.
BUT: The development and maintenance cost would have been extreme, because we don't yet use C++14 with relaxed constexpr rules. This also doomed the SystemClock PR, since complex C++11 constexpr become unmaintainable after a while.

So, let's not bash everything here, we are beating even newlib by a mile (I'm not even talking about vendor code anymore).

ekiwi, 2016-05-08:
> So, let's not bash everything here

It was not my intention to "bash" anything. Currently you are the most active developer and thus most recent changes will be introduced by you.
I just think that link time optimizations are something that should be worth investigating, as they have been a feature of gcc for a long time now and promise size reduction improvements.

salkinium, 2016-05-08:
Sry, I wasn't offended or directing my comment at anything particular, I just don't want us to get overexcited about the optimization potential of flash size, since xpcc is already pretty small.

closed 2016-05-08 by salkinium

salkinium, 2016-05-08:
@ekiwi You also need to pass the `-lto` flag to the linker.
However, the `binutils` that ship with the precompiled arm-none-eabi-gcc package are not compiled using `--enable-lto` and therefore fail with the message "plugin needed to handle lto object".

salkinium, 2016-05-08:
@ekiwi Passing `-ffat-lto-objects` to gcc solves the issues above and drops size to 1284B Flash + 8B static RAM. PR is coming.

PR #123 (2016-05-08): Stop TLSF using newlib's assert and printf.
This accidentally added ~30kB of code and ~2kB of static RAM usage.

Fixes #122.