roboterclubaachen/xpcc #132: Outdated VBoxGuestAdditions

Jinzuu, 2016-05-17:
Hi,

if you try to run the Vagrant-Box with a newer VBox version (mine is currently 5.0.18), you might run into trouble when trying to pass-through the STM32-ST-Link.

There should be a section in the Getting started mentioning how to upgrade the VBoxGuestAdditions.
In my case it sufficed to run in the VM:
`sudo apt-get update`
`sudo apt-get install --only-upgrade virtualbox-guest-utils`

salkinium, 2016-05-17:
Thanks for testing that!

I've rebuild the VM today, as the one available was one year old.
However, there is some outdated software, that needs updating as well (like openocd 0.10.0, arm-none-eabi-gcc 5, etc), so I'm working on that.

I'll update the documentation in one go with your suggestions, once I can test this with a good VM.

salkinium, 2016-05-23:
Some research pointed me to [the `vagrant-vbguest` plug-in](http://kvz.io/blog/2013/01/16/vagrant-tip-keep-virtualbox-guest-additions-in-sync/) which manages guest additions automatically.
That's pretty cool!

salkinium, 2016-11-20:
The VM is not supported anymore, due to maintenance issues.

closed 2016-11-20 by salkinium