roboterclubaachen/xpcc #133: Getting started for USB integration with Vagrant Headless-Box

Jinzuu, 2016-05-17:
There is no section about how to pass-through the ST-Link to the Vagrant Box.
After the Installation & Compilation section, it just says that you can type `scons program` to program. But at that point, there is no ST-Link connected, yet.

I found this solution by merging some tutorials on the internet:
1. Update the VBoxExtensions
2. 
Inside the Vagrantfile, after the lines
```
config.vm.box = "xpcc"
config.vm.box_url = "http://box.xpcc.io/xpcc-vm.box"
```
Add this 
```
config.vm.provider "virtualbox" do |vb|
    vb.customize ["modifyvm", :id, "--usb", "on"]
    vb.customize ["modifyvm", :id, "--usbehci", "off"]
  end

  config.vm.provider "virtualbox" do |vb|
    vb.customize ["usbfilter", "add", "0",
        "--target", :id,
        "--name", "Any STM32",
        "--product", "STM32 STLink"]
  end
```
This attaches **any** ST-Link to the Box.

salkinium, 2016-05-17:
Yes, good point. That documentation is not clear at all!
It doesn't even address the VBoxExtensions. 😞 

I gave your solution a try and the `usbfilter add ...` seems to add its filter on every `vagrant up`, which is suboptimal.
Also with my VBox v5.0.20, the `--product "STM32 STLink"` is ignored, not sure why.

I'll investigate further.

salkinium, 2016-05-23:
I added this [to the template of the VM](https://github.com/roboterclubaachen/rca-vm/commit/f706986b285c0a006d0aceee4af28a32b418b408), this way the filter won't be added on every boot.

salkinium, 2016-11-20:
I couldn't find a reliable USB connection way for VirtualBox, so the VM is not supported anymore, also due to maintenance issues.

closed 2016-11-20 by salkinium