roboterclubaachen/xpcc #136: TLSF allocator breaks program

Sh4rK, 2016-05-18:
Hi!

I updated xpcc to the latest version recently for the I2C fix, but unfortunately my program stopped working properly, nothing is drawn on the screen nor do input buttons work. I used git bisect to find that the breaking commit is 0b457d65525a9ad8f50bd99d8ec961e8ae807c9b, which enables the TLSF allocator, but I don't use dynamic allocation at all. I didn't get to debugging yet (It's just so cumbersome to set up on windows), but from experimenting I suspect something in xpcc, more specifically in formatting (probably floats) might be the cause.

I'll comment when I know more.

salkinium, 2016-05-18:
I shouldn't have enabled TLFS by default, I'll revert it.

I don't think TLSF is at fault, but the way we configured newlib is the root issue.

Sh4rK, 2016-05-18:
Manually setting the allocator back to newlib solved the issue, but I still don't know the exact cause, and I don't have time right now to find out :/

salkinium, 2016-05-18:
I've renabled newlib's allocator for now, since this is too critical to leave be.

This issue stays open until this is properly fixed.

PR #147 (2016-06-22): Add memory traits for dynamic memory using TLSF
This adds the concept of memory traits to the allocators, incl. global overloading of operator `new`.
This expands the work in #115 by allowing the user to request memory with certain traits, like DMA-able, Fast access, or external RAM.

All heap arenas are now stored in an array constructed in the linker script, which is used to initialize all allocators at runtime. If TLSF is not the chosen allocator, then newlib is initialized with the largest continous heap arena, that satifies the `xpcc::MemoryDefault` trait (ie. at least DMA-able and S-Bus  [... cut]
closed 2016-06-22 by salkinium