roboterclubaachen/xpcc #150: Support for Atmel SAM D21

joseangeljimenez, 2016-06-24:
Do you plan to support Atmel SAM D line of microcontrollers (ARM Cortex M0+ based),
http://www.atmel.com/products/microcontrollers/arm/sam-d.aspx

Thanks for your great work!

salkinium, 2016-06-24:
I think it may be relatively easy to hack something very basic together, since the [ARM Cortex-M core](https://github.com/roboterclubaachen/xpcc/tree/develop/src/xpcc/architecture/platform/driver/core/cortex) stuff (startup, linkerscript) is highly similar, but porting the HAL is a lot of work.

ST has a bit of an advantage, since they publish a lot of machine readable data, which helps a lot with the HAL porting. I haven't found similar data for Atmel chips :-(

So realistically: no, there are no plans. We simply don't have enough developers to deal with this now or in the conceivable future.

joseangeljimenez, 2016-06-24:
Thanks for your prompt response. I will try to hack something basic together, as you suggest, as see how far I can move from there on.

salkinium, 2016-06-24:
Since we lack a porting guide (fail!), I recommend looking at [the minimal LPC11 port](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/devices/lpc/lpc11c24.xml) we've done, which is Cortex-M0 (not M0+ though).
Also see [a better STM32F0 port](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/devices/stm32/stm32f072-c_r_v-8_b.xml#L17-L53), which are also CM0.

For an absolute minimal port you need to get [`platform/driver/core`](https://github.com/roboterclubaachen/xpcc/tree/develop/src/xpcc/architecture/platform/driver/core/cortex)(the [`startupCode` macro](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/lpc/lpc.macros#L46), [`xpcc_gpio_enable()`](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/startup.c.in#L290) requires implementation, use [stm32_ram.ld](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/stm32/stm32_ram.ld.in)) and [`platform/driver/gpio`](https://github.com/roboterclubaachen/xpcc/tree/develop/src/xpcc/architecture/platform/driver/gpio) running, which gives you bitbanged [SPI](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/spi/generic/spi_master.hpp#L36) and [I2C](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/i2c/generic/i2c_master.hpp#L33).
I strongly recommend getting [`platform/driver/uart`](https://github.com/roboterclubaachen/xpcc/tree/develop/src/xpcc/architecture/platform/driver/uart) running as well, so you get some serial output as well (but for blinky it's not necessary).

ekiwi, 2016-06-25:
> Thanks for your prompt response. I will try to hack something basic together, as you suggest, as see how far I can move from there on.

Feel free to ask questions on this issue or via our [mailing list](http://mailman.rwth-aachen.de/mailman/listinfo/xpcc-dev). I did a lot of the basic HAL stuff and I will try to help you as much as possible.

Can you tell us, what exact hardware you have available?

joseangeljimenez, 2016-07-03:
It is a custom board based on the ATSAMD21, the version with 128KB of Flash memory. Still on the drawing board... :-)

ekiwi, 2016-07-03:
> It is a custom board based on the ATSAMD21

Ok. If you want to port `xpcc` I would recommend you first use an official prototyping board, so that you can be sure that the hardware is working correctly, when you try to debug your software.

Depending on your application, if you want to use `xpcc`, you could also have a look at some STM32 processors as we have support for a lot of them and it is relatively easy to add support for the ones that we are missing at the moment.

For example we have decent support for the STM32F072 which is a cortex-m0 base microcontroller with CAN and USB (we do not have USB software support at the moment).
Or you can use the older STM32F103 microcontroller for which cheap Chinese boards are available on eabay. There is a [pull request](https://github.com/roboterclubaachen/xpcc/pull/154) which adds support for this board.
If you need a cortex-m0+ based controller it should be possible to add support for the STM32L0 line in `xpcc` with much less effort than it would be to support the SAMD21.

If you chose to stay with the ATSAMD21 we would of cause still be interested in supporting it as a new target. If you want you can drop us an email if you need help.

If you can tell us a little bit more about your project or your programming background, @salkinium and I can help you figure out what is supported in `xpcc` and what would need to be developed in order for your project to run on `xpcc`.

joseangeljimenez, 2016-07-19:
We are still working at the board level, on various prototypes, all of them based on the ATSAMD21E variant (32 pin, various options for Flash/DRAM capacity). It is not feasible, for now, switching to another hardware/MCU architecture. My background is in EE, with years of experience in hardware design and FW development in C++, however, I am relatively new to ARM.

Thanks for your support. I may come back later when having started doing some work and tried your framework a little bit more.

ekiwi, 2016-07-20:
> My background is in EE, with years of experience in hardware design and FW development in C++

Sounds like `xpcc` could be fitting for you. We make use of a lot of cool C++ features and try to stay true to our EE backgrounds regarding efficiency and memory allocations.

> Thanks for your support. I may come back later when having started doing some work and tried your framework a little bit more.

OK. Just ask if there is anything unclear.

ekiwi, 2016-09-03:
@joseangeljimenez any updates on your porting efforts?
xpcc is about to have a [porting guide](https://github.com/roboterclubaachen/xpcc/pull/179) which you might find useful.

joseangeljimenez, 2016-09-05:
Hello. For the time being I had to move forward with another framework. However, I do not discard to invest some time to do the porting in the future.

Thanks for your support!

closed 2016-09-05 by joseangeljimenez