roboterclubaachen/xpcc #159: LPC11 GCC linker crash

strongly-typed, 2016-07-02:
Iterating all the compiles in [TravisCI](https://travis-ci.org/roboterclubaachen/xpcc/jobs/141835097#L1042) reveals some incompatibility. Still works with [2015q2](https://travis-ci.org/roboterclubaachen/xpcc/jobs/141835154#L1046). New is always better ;-)

I have the same problem with 2016q2 on OS X.

salkinium, 2016-07-02:
```sh
Linking: /home/travis/build/roboterclubaachen/xpcc/build/lpcxpresso/lpc11c24/uart/uart.elf
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
```
Dunno what to say to that. Maybe ¯\\_(ツ)_/¯ ?

I've seen this before (ld currently crashes on our internal CI too), and it seems to have something to do with putting certain things into the linkerscript. That's literally as much as I know. Just changing one line, or moving sections around will fix this, but I would still like to know exactly what the problem is. This can be extremely frustrating to debug.

strongly-typed, 2017-03-18:
PR #225 "resolves" this issue by using the latest 2017q1 ARM GCC which does not crash.

closed 2017-03-18 by strongly-typed