roboterclubaachen/xpcc #169: Else Statement Considered Harmful / STM32F1 PWM broken

strongly-typed, 2016-08-14:
A STM32F103 blue pill + xpcc + some lines of code = PWM?

What started as a very simple task ended up in a big misery :-(

	Timer1::enable();
	Timer1::setMode(Timer1::Mode::UpCounter);
	Timer1::setOverflow(1000);
	Timer1::applyAndReset();
	Timer1::start();

	GpioOutputA8::connect(Timer1::Channel1);
	Timer1::configureOutputChannel(/* channel = */ 1, Timer1::OutputCompareMode::Pwm, /* compareValue = */ 500);

	Timer1::enableOutput();

This should produce a nice square wave at GPIO A8, but it didn't.

After debugging every single timer register I found out the the GPIO is not configured properly by`GpioOutputA8::connect(Timer1::Channel1);`

The code produced in `gpio.hpp` is

	// connect: Hardware Gpio
	/// Connect GpioOutputA8 as Channel1 to Timer1. OOMMGGGGdddd 
	ALWAYS_INLINE static void
	connect(TypeId::Timer1Channel1 /* t */) {
		configure(InputType::Floating);
		AFIO->MAPR = (AFIO->MAPR & ~(3 << 6)) | (0 << 6);
	}

which is generated from the [gpio.hpp.in](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/gpio/stm32f1/gpio.hpp.in#L318)

	/// Connect {{name}} as {{af.name}} to {{af.peripheral}}.
	ALWAYS_INLINE static void
	connect(TypeId::{{ af.peripheral }}{{ af.name }} /* t */) {
					%% if af.type == "analog"
		setAnalogInput();
					%% else
						%% if af.type == 'out'
		configure(OutputType(i(OutputType::PushPull) | 0x8), OutputSpeed::MHz50);
						%% else
		configure(InputType::Floating);
						%% endif
		{{ enableAlternateFunction }}
					%% endif
	}
				%% endif
				%% if af.type|string in ['', 'out'] and type != "Input" and af.name not in ['Scl', 'Sda']

The `configure(InputType::Floating)` is added because the substitution dictionary contains 

     {'id': '8', 'port': 'A', 'afs': [
          {'peripheral': 'Timer1', 'id': '6,3,0', 'name': 'Channel1'}, 
          {'peripheral': 'ClockOutput', 'type': 'out'}, 
          {'peripheral': 'Uart1', 'type': 'out', 'name': 'Ck'}, 
          {'peripheral': 'UartSpiMaster1', 'type': 'out', 'name': 'Sck'}
     ]}

And that's because [stm32f103-c_r_t_v-8_b.xml](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/devices/stm32/stm32f103-c_r_t_v-8_b.xml#L134) is missing

     type="out"

for all Timer Output Channels.

I doubt that the PWM ever worked on a STMF1.

IMHO `else` is Considered Harmful because it adds a default case which is not intended. This is no self-healing code. I would suggest (like the enum class case without default)

     if out then
          out
     else if in then
          in
     else
          ERROR unexpected

That's the same with the `if target == stm32f4 else` statements. The else branch unintentionally covers all other platforms without explicit consent of the author. I would suggest to eliminate that use of `else` where possible.

For my own reference:

[platform_tools.py](https://github.com/roboterclubaachen/xpcc/blob/develop/scons/site_tools/platform_tools.py#L195) let [template.py](https://github.com/roboterclubaachen/xpcc/blob/develop/scons/site_tools/template.py#L57) generate `gpio.hpp` from [gpio.hpp.in](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/gpio/stm32f1/gpio.hpp.in). The `substitutions` dictionary is generated by [driver.py](https://github.com/roboterclubaachen/xpcc/blob/develop/tools/device_files/driver.py#L96) from [stm32f103-c_r_t_v-8_b.xml](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/devices/stm32/stm32f103-c_r_t_v-8_b.xml).

salkinium, 2016-08-14:
Ohai there with the catchy headline.

> I doubt that the PWM ever worked on a STMF1.

I doubt that any of the Timer code was ever really tested on all platforms. Welcome to the test team ;-P

This particular issue is very interesting because it touches the entire generation stack.
First there was a failure in parsing the raw data and generating the device file, which is missing `type="out"` for all output channels. Second, there was a failure in verifying the information in the template. And third, it was never tested for functional compliance.

I think you are channeling your anger on the wrong thing. Of course `else` is not considered harmful, I think you just like that headline. Yes, it caught my attention.
A different design wouldn't have helped, because the AF is missing the `type` data. This signifies that the AF is valid for both input and outputs. If we added an explicit `inout` type, this wouldn't change anything, since the data in the device file then would still simply be wrong. There is nothing the template can do to "heal" wrong input. It's also the wrong piece of the stack to deal with handling wrong input.

The choice of using `else` in the HAL templates boils down to maintainability when new devices are added. Specifying every target explicitly quickly becomes a nightmare. Look at the [choice matrix for the linkerscript](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/driver.xml#L37-L38) in our (shitty) `driver.xml` "language", which is missing `else`. I really don't want that explicitness in every driver file, that'll be horrible to maintain.

I would actually prefer to pull as much target specific tests out of the template and into the `driver.xml`, so that you only have to update one file per driver. At the moment, the `driver.xml` format it shitty, we should just replace it with a regular python file that add variables to the substitution dictionary. All `if target == stm32f4 else` would then be replaced by a variable, that describes the template specialization based on functionality rather than the target directly. Porting to the STM32L4 is a little painful, because you have to add the `target is stm32l4` to (almost) every line of `target is stm32f4`. That approach does not scale well.

So this issue is really quite layered in architecture design decisions and could have been mitigated by a device file with *correct* data or perhaps even a device file verification step.
However, I think our HAL generation build system is to blame, with a lot of it being in a state of terrible hackyness (just think about the parameters) and the lack of support for pulling the target specificness out of the the HAL templates themselves.

Oh, and since I'm in a sarcastic mood, here is another gem about our device file generator: The F1 alternate function mechanism is so special that it is not described in ST's cube files. I had to add [this information manually to the device file generator](https://github.com/roboterclubaachen/xpcc/blob/develop/tools/device_file_generator/stm.py#L93-L119). Have all combinations of this on all chips been tested? Heck no.

Even better. I had to add [all GPIOs alternate functions of all AVRs manually](https://github.com/roboterclubaachen/xpcc/blob/develop/tools/device_file_generator/avr_io.py), because no such data exists in machine readable form. I had to open all datasheets and put it in a massive Excel sheet. Of course I made mistakes there. Has it been tested on all AVRs? Heck no.

Hope that makes you feel better 😝.

But in all seriousness, this is a symptom of bigger issue with the way we generate the HAL with our build system. Turning all `else` into explicit "switches" is [classic bikeshedding](https://en.wiktionary.org/wiki/bikeshedding), and doesn't solve the real issue. Doesn't mean we shouldn't fix this ASAP of course!

salkinium, 2016-08-14:
More on this specific problem: The lack of `type="out"` in the device files seems to be intentional, since [the reader explicitly removes it](https://github.com/roboterclubaachen/xpcc/blob/develop/tools/device_file_generator/stm_reader.py#L379).
On the non-F1 GPIO template, this is not an issue, since [only `type="analog"` is tested for](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/gpio/stm32/gpio.hpp.in#L409-L413). 

I therefore believe this is an issue of the template rather than the device file data.

strongly-typed, 2016-08-14:
> I doubt that any of the Timer code was ever really tested on all platforms. Welcome to the test team ;-P

[I'm in!](https://github.com/roboterclubaachen/xpcc/commit/a0c109e08c20f360bb80f902509fabcdebe6e70d#diff-7de10eade98a12452702c93bad683669)

salkinium, 2016-09-07:
I reverted #172 since it broke valid code for non-F1 platforms. See #178 for discussion.

salkinium, 2019-01-06:
This has been fixed in modm.

PR #172 (2016-08-19): [DFG] Add `out` type to all STM32 timer channels.
Fixes #169.

cc @strongly-typed
closed 2019-01-06 by salkinium