roboterclubaachen/xpcc #171: Gmane email list archive is dead

salkinium, 2016-08-17:
The xpcc mailing list is still available of course, being hosted by the RWTH Aachen University, which also has a [less usable archive here](http://mailman.rwth-aachen.de/pipermail/xpcc-dev/).

We need to update the link in the Readme.

ekiwi, 2016-08-17:
What are we going to do about that?

It's nice to have our emails archived somewhere accessible to google.
RWTH Aachen has configured their `robot.txt` in a way that blocks google from indexing.

salkinium, 2016-08-17:
> What are we going to do about that?

"Ingebrigtsen recommends people to have a look at [Mail Archive](http://www.mail-archive.com)."

Example look for the [avr-gcc mailing list](https://www.mail-archive.com/avr-gcc-list@nongnu.org/maillist.html). Not as pretty as Gmane, but it'll do I guess.

ekiwi, 2016-09-03:
Are you the administrator of our mailing list @salkinium?
If so, could you please [subscribe the Mail Archive](https://www.mail-archive.com/faq.html#newlist) to our list?

salkinium, 2016-09-05:
I've added the mail archive as a subscriber. There's no page yet, cos no email has been sent over the list yet.

salkinium, 2016-09-05:
Oh, never mind, it just works:
https://www.mail-archive.com/xpcc-dev@lists.rwth-aachen.de/

PR #181 (2016-09-05): Update link to mail archive.
gmane.org is dead, long live mail-archive.com. Closes #171.

Will update for xpcc.io website as well.

cc @ekiwi (just FYI)
closed 2016-09-05 by salkinium