roboterclubaachen/xpcc #177: src/xpcc/ui/gui/widgets/widget.cpp:46:1: warning: all paths through this function will call itself [-Winfinite-recursion]

strongly-typed, 2016-08-28:
[src/xpcc/ui/gui/widgets/widget.cpp:46:1:](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/ui/gui/widgets/widget.cpp#L47) warning: all paths through this function will call itself [-Winfinite-recursion]

What's wrong  here?

ekiwi, 2016-08-28:
nice find!

This was most likely introduced in 447ac90ae3afee3a04bd764ebde7853ebe10b5cd.

Reminds me of this [Debian cleanup](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=363516)

ekiwi, 2016-08-28:
@daniel-k, @salkinium  can you try to figure out what the future of our (somewhat broken) GUI stack will be?
Maybe there are some new students at RCA that want to start developing for `xpcc` and would like to extend and clean up the GUI stuff.

salkinium, 2016-08-28:
So obviously [`salkinium/upainter`](https://github.com/salkinium/upainter) is my embedded graphics research project, but I didn't find the motivation to continue working on it. It's missing all code for the widgets, since I wanted to understand graphics first. (Also it requires C++14 and didn't run on an embedded device yet).

I believe that the current solution is sufficient for our simple GUI needs.
The new™ RCA display _could_ now be implemented using the newly added support for the STM32F469 display and touchscreen (#155), though that may require some hardware effort for adding all the features from the current display, and I think we can reuse the displays from last year.

It all boils down to people taking ownership of something and writing code. That's why we even have a GUI system at all.

ekiwi, 2016-08-28:
> I believe that the current solution is sufficient for our simple GUI needs.

OK. I just feel like the whole architecture involving the color palette is fundamentally broken: First I tried to make it compile on AVR and then you "fixed" it for clang and produced an infinite loop. So I feel like there is no quick fix for this issue, but someone (TM) has to reevaluate the design and figure out how to implement the color palette in a clean and non broken way.

salkinium, 2016-08-28:
Well, I tried to "do it right" once and now I'm yak-shaving upainter…

ekiwi, 2016-08-28:
> Well, I tried to "do it right" once and now I'm yak-shaving upainter…

We do not change out complete model. I just think that the way `ColorPalette` is implemented in fundamentally broken.

salkinium, 2016-08-28:
> I just think that the way ColorPalette is implemented in fundamentally broken.

[image]