roboterclubaachen/xpcc #178: [DFG] `in` type is missing for STM32 timer channels.

strongly-typed, 2016-08-28:
In #172 `out` was added. But this disables the use of the timer pin as [encoder input](https://github.com/strongly-typed/xpcc/blob/feature/zmq/examples/zmq/1_stm32/main.cpp#L64).

salkinium, 2016-08-29:
So adding `out` to all channels was obviously the wrong solution, because now the AF is only available for non-input pins.
Sigh, I wish I had a brain…

salkinium, 2016-08-29:
So the issue is this the hardware for the F1, which isn't mappable with the current API design:
```cpp
						%% if af.type == 'out'
		configure(OutputType(i(OutputType::PushPull) | 0x8), OutputSpeed::MHz50);
						%% else
		configure(InputType::Floating);
						%% endif
		{{ enableAlternateFunction }}
```

On all other families, the AF configures pin direction, but on the F1 it has to be explicitly set.
However, xpcc cannot know if `Gpio::connect(Timer::Channel[0-3]);` means input or output use.

The underlying issue is of course that we don't actually have a Timer interface, because we've never established clear use-cases of what we wanted to support.
So this is an area which design has simply not been finished.

If I look into the device file, there are several AF signals for timers:
- `Channel[0-9]`, or `Channel[0-9]N` (inverted? negated?)
- `ExternalTrigger`
- `BreakIn`

Unfortunately I don't actually know what `BreakIn` is useful for, and only have a superficial understanding of `Channel[0-3]N` and `ExternalTrigger`.

What would a useful interface for Timer AFs look like?
Adding a second parameter with the channels use?
`Gpio::connect(Timer::Channel[0-3], Timer::Use::Pwm)` versus
`Gpio::connect(Timer::Channel[0-3], Timer::Use::Encoder)`?
Perhaps flatten the matrix? 
`Gpio::connect(Timer::Channel[0-3]Encoder)`?

strongly-typed, 2016-08-29:
> Channel[0-9]N (inverted? negated?)

Yes, inverted. It can be used for complementary PWM with dead-time insertion. Think of H-bridges and the high and low-side MOSFETs.

strongly-typed, 2016-08-29:
RM0008, 14.2

> Break input to put the timer’s output signals in reset state or in a known state.

salkinium, 2016-09-06:
I had to reverted the commit that introduced this "bug" in https://github.com/roboterclubaachen/xpcc/commit/6d355b655878ba1134362366d2d4d845e0a54fc1, since it broke our robot code and that has highest priority.

As stated above, the current API is broken from a use-case PoV and needs to be reworked.
However, it is questionable whether the `out` information should even be part of the device files in the first place, since it makes assumptions on API usage, specifically that there is a static distinction between an output and an input, which is xpcc/C++ specific.

salkinium, 2016-10-02:
This issue is worse than it looks. All GPIO AF inside the device files for the F1 series is ambiguous resulting in wrong matches.
The issue is that pins can only be remapped in groups, however, those groups are overlapping each other, which is not mappable to the current data format in the device files and especially not mappable to individual `connect` methods (without keeping some kind of state).

The assumptions we made when designing the `connect` method were true for F4, but they are not true for F1 and ATxmega and in some special cases also not even true for ATmega.
They all remap groups of pins!

Fixing this requires coupling the Pins to the peripheral, in the same way we already do for `xpcc::SoftwareI2cMaster<...>` and `xpcc::SoftwareSpiMaster<...>`. The `connect` method disappears into the peripheral then, which also fixes the issue of the order dependency of some `connect` methods (like connecting I2C SCL).

This would not require rewriting all of HAL, just add a thin layer on top of it, which deals with pins. It would look like this:
```cpp
// Usart can be used for UART
using LogUart = xpcc::stm32::Usart2::Uart<Tx, Rx>;
// But also for SPI
using DisplaySpi = xpcc::stm32::Usart2::SpiMaster<Sck, Mosi, Miso>;
// Future SpiSlave
using CommSpi = xpcc::stm32::Usart2::SpiSlave<Sck, Mosi, Miso, Nss>;

// somewhere in main
// no more connect methods!
LogUart::initialize<systemClock, 115200>();
```

The actual code that connects the GPIOs is delegated to the GPIO driver, by template overloading. The TypeIDs are now "reversed", and do not apply per pin, but per peripheral. The Uart driver above would in it's `initialize` method call:
```cpp
xpcc::stm32::Gpio::connect<Tx, Rx, Uart2>();
```
which is an specialized template (for groups) or a constexpr function (for crossbar implementations) that throws a compile error, if the connection does not exist.

Note that all templated Uart implementations can use the same Uart HAL implementation, that does not know about the pins. This will keep code bloat at bay, but if it is absolutely required that the HAL knows about the connect pin, it may choose to make the implementation dependent on it.

salkinium, 2016-11-25:
Short update: I haven't forgotten about this. I've [extracted the correct PinMux information from the ST data and added them to the device files](https://github.com/modm-io/modm-devices/commit/5a4e4b9bff2cdd300468f3118ec5d53c048d7095). 
Be aware that these changes require build system support that xpcc does not have, but modm/lbuild does. It therefore won't be backported to xpcc.

The `connect()` method for STM32F1 is therefore not supported in xpcc. I plan to add a manual pinmux HAL as a temporary fix until modm is ready.

salkinium, 2017-06-05:
This has been successfully implemented in modm for all remaps with the same API for all devices.
For [example on the NUCLEO-F103RB](https://github.com/modm-io/modm/blob/develop/src/modm/board/nucleo_f103rb/nucleo_f103rb.hpp#L118), connecting Uart currently works like this:
```
GpioConnect<Peripheral::Usart2, GpioOutputA2::Tx, GpioInputA3::Rx>::connect();
```
Connecting pins of conflicting groups results in a static assertion failure:
```
GpioConnect<Peripheral::Usart1, GpioA9::Tx, GpioB7::Rx>::connect();
```
[image]
Or a more complex remap group:
```
GpioConnect<Peripheral::Timer2, GpioA0::Ch1, GpioB3::Ch2>::connect();
```
[image]

Note that `GpioConnect` is an internal class, which only connects the AFs of the pin, *it does not set up the pin configuration*!
It is supposed to be called by the driver in a `connect()` method (for example: `Usart1::Uart::connect<GpioA2::Tx, GpioA3::Rx>();`) which then first sets up the pin configuration before connecting. This way, the I2cMaster can strobe the Scl pin before connecting it.

This API won't be backported to xpcc, since it requires a [significant amount of data processing](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/stm32/module.lb), that would be hard to implement in the xpcc build system.
cc @strongly-typed

salkinium, 2017-06-07:
> It is supposed to be called by the driver in a connect() method

This has been prototyped in modm now for [Uart](). We're using [`GetSignal<Signal>`](https://github.com/modm-io/modm/blob/wip/niklas/src/modm/platform/gpio/stm32/gpio_connect.hpp.in#L26-L42) to extract the right pins from the pin group, then configuring them as the driver needs them.
If a pin doesn't exist in the connector pin group, [`stm32::GpioUnused`](https://github.com/modm-io/modm/blob/wip/niklas/src/modm/platform/gpio/stm32/gpio_unused.hpp.in#L58-L119) is used instead which just ignores all Gpio commands.
This is therefore an "optional default initialization" procedure for connected pins, ie. it can be called as `Usart1::connect<GpioA2::Tx>();`, `Usart1::connect<GpioA3::Rx>(Gpio::InputType::PullUp);` or `Usart1::connect<GpioA2::Tx, GpioA3::Rx>(Gpio::InputType::PullUp);` and in all cases it works as expected:
```c++
template< template<Peripheral _> class... Gpios>
static void
connect(Gpio::InputType RxInputType = Gpio::InputType::PullUp)
{
    using Connector = GpioConnect<Peripheral::{{ name }}, Gpios...>;
    using Tx = typename Connector::template GetSignal< Gpio::Signal::Tx >;
    using Rx = typename Connector::template GetSignal< Gpio::Signal::Rx >;

    Tx::setOutput(Gpio::OutputType::PushPull);
    Rx::setInput(RxInputType);
    Connector::connect();
}
```

The [I2cMaster::connect](https://github.com/modm-io/modm/blob/wip/niklas/src/modm/platform/i2c/stm32/i2c_master.hpp.in#L100-L114) is a bit more involved due to the `resetDevices<Scl>()` function. Here both the Scl and Sda are required to be connected at the same time, otherwise the driver cannot work. Calling `I2cMaster::connect<GpioB3::Scl>()` will fail with a static_assertion failure.

```c++
template< template<Peripheral _> class... Gpios>
static void
connect(bool with_pullup = false, bool reset_devices = true)
{
    using Connector = GpioConnect<Peripheral::I2c{{ id }}, Gpios...>;
    using Scl = typename Connector::template GetSignal< Gpio::Signal::Scl >;
    using Sda = typename Connector::template GetSignal< Gpio::Signal::Sda >;
    static_assert(not typename tmp::SameType<Scl, GpioUnused>::value and
                  not typename tmp::SameType<Sda, GpioUnused>::value,
                  "Both Scl and Sda need to be connected!");

    Scl::configure(with_pullup ? Gpio::InputType::PullUp : Gpio::InputType::Floating);
    Sda::configure(with_pullup ? Gpio::InputType::PullUp : Gpio::InputType::Floating);
    Scl::setOutput(Gpio::OutputType::OpenDrain);
    Sda::setOutput(Gpio::OutputType::OpenDrain);
    if (reset_devices) resetDevices<Scl>();
    Connector::connect();
}
```

This works the same on **all** STM32 and AVRs, especially with the group remaps of STM32F1/L1 and Xmega.

ekiwi, 2017-06-07:
Can you link the generator scripts from `modm`?
Maybe there is a way to backport this. I cannot promise anything but would like to have a look at it.

salkinium, 2017-06-07:
> Can you link the generator scripts from modm?

They are generated from [modm-devices/wip/niklas](https://github.com/modm-io/modm-devices/tree/wip/niklas). To reproduce:
```zsh
cd tools/generator
make extract-data-avr
make extract-data-stm32
make generate-avr
make generate-stm32
```
It's very, very WIP, so you may have to adapt some flags for your shell (I think `ack` is OS X only?). STM32 is fine, but for the AVR I only have GPIO signal information for about ⅓ of the devices. Atmel simply doesn't provide it.

> Maybe there is a way to backport this

Please please don't, I'm already ~punished~ lucky enough to forward-port all the changes from xpcc to modm, I'm _really not_ gonna do that the other way too. 
Let's move modm forward instead. There are a lot of other fixes that the new build-system enabled.

salkinium, 2017-06-07:
> Can you link the generator scripts from modm?

On second read, I think you wanted the `modm:platform:gpio` module for [STM32](https://github.com/modm-io/modm/tree/wip/niklas/src/modm/platform/gpio/stm32) and [AVR](https://github.com/modm-io/modm/tree/wip/niklas/src/modm/platform/gpio/at90_tiny_mega).

salkinium, 2019-01-06:
This has been fixed in modm.

closed 2019-01-06 by salkinium