roboterclubaachen/xpcc #184: Compiler issues with GCC 6.1+

daniel-k, 2016-09-12:
The F072 and F3 rotation example has the following problem:

~~~
main.cpp: In member function 'bool ReaderThread::update()':
main.cpp:53:15: error: invalid operands of types 'int' and '__gnu_cxx::__enable_if<true, double>::__type {aka double}' to binary 'operator<<'
     leds = (1 << abs(leds)) - 1;
             ~~^~~~~~~~~~~~
~~~

With the corresponding code:
~~~c
uint16_t leds = abs(value / 200 * 5);
leds = (1 << abs(leds)) - 1;
~~~

Why is the second `abs()` needed at all? `leds` is even unsigned, so I don't see any need.

Moreover, all lpc examples fail with the following error:
~~~
[...]/xpcc/build/lpcxpresso/lpc11c24/blink/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/can/lpc/c_can.hpp:149:14: error: reinterpret_cast from integer to pointer
  ROM **rom = (ROM **) 0x1fff1ff8;
              ^~~~~~~~~~~~~~~~~~~
~~~

ekiwi, 2016-09-12:
> Why is the second abs() needed at all? leds is even unsigned, so I don't see any need.

it seems to be redundant

> Moreover, all lpc examples fail with the following error:

Hehe. Evil C-style casts!

salkinium, 2016-09-20:
Fixed in https://github.com/roboterclubaachen/xpcc/commit/b72e255f27656d29346e942d2736690d2cea100b and https://github.com/roboterclubaachen/xpcc/commit/6e3360621c0966be6e9c44953c51387f7d79fa42.

closed 2016-09-20 by salkinium