roboterclubaachen/xpcc #192: [ci] Messing with paths in Scons

strongly-typed, 2016-10-01:
I'm still trying to track down the `import jinja2` problem and found out that the path differs in different scripts.

Any clues where the sys.path is messed up?

On my OS X:

    sys.path in builder_base.py: [
    '/Users/usr/Dev/xpcc_travis_ci_trusty/tools/system_design/builder', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python27.zip', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', 
    '/usr/local/lib/python2.7/site-packages', 
    '/Library/Python/2.7/site-packages'
    ]
    
    sys.path in jinja2_template_action: [
    '/usr/local/bin/../engine', 
    '/usr/local/Cellar/scons/2.5.0/libexec/scons-local-2.5.0', 
    '/usr/local/Cellar/scons/2.5.0/libexec/scons-local', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/scons-2.5.0', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons-2.5.0', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/scons-2.5.0', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/scons', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/scons', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/scons', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python27.zip', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', 
    '/usr/local/Cellar/python/2.7.12_1/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', 
    '/usr/local/lib/python2.7/site-packages', 
    '/Library/Python/2.7/site-packages'
    ]

In Travis CI:

`https://travis-ci.org/strongly-typed/xpcc/builds/164302450#L1720`

	sys.path in builder_base.py: [
	'/home/travis/build/strongly-typed/xpcc/tools/system_design/builder', 
	'/opt/python/2.7.10/lib/python27.zip', '/opt/python/2.7.10/lib/python2.7', 
	'/opt/python/2.7.10/lib/python2.7/plat-linux2', 
	'/opt/python/2.7.10/lib/python2.7/lib-tk', 
	'/opt/python/2.7.10/lib/python2.7/lib-old', 
	'/opt/python/2.7.10/lib/python2.7/lib-dynload', 
	'/opt/python/2.7.10/lib/python2.7/site-packages'
	]

`https://travis-ci.org/strongly-typed/xpcc/builds/164302450#L1730`

	sys.path in jinja2_template_action: [
	'/usr/bin/scons-local-2.3.0', 
	'/usr/bin/scons-local', 
	'/usr/lib/scons-2.3.0', 
	'/usr/lib/scons-2.3.0', 
	'/usr/local/lib/scons-2.3.0', 
	'/usr/lib/python2.7/site-packages/scons-2.3.0', 
	'/usr/lib/python2.7/site-packages/scons-2.3.0', 
	'/usr/local/lib/python2.7/site-packages/scons-2.3.0', 
	'/usr/lib/scons-2.3.0', 
	'/usr/lib/scons', 
	'/usr/lib/scons', 
	'/usr/local/lib/scons', 
	'/usr/lib/python2.7/site-packages/scons', 
	'/usr/lib/python2.7/site-packages/scons', 
	'/usr/local/lib/python2.7/site-packages/scons', 
	'/usr/lib/scons', 
	'/usr/lib/python2.7', 
	'/usr/lib/python2.7/plat-x86_64-linux-gnu', 
	'/usr/lib/python2.7/lib-tk', 
	'/usr/lib/python2.7/lib-old', 
	'/usr/lib/python2.7/lib-dynload', 
	'/usr/local/lib/python2.7/dist-packages', 
	'/usr/lib/python2.7/dist-packages'
	]

salkinium, 2016-10-01:
cc @ekiwi @dergraaf

dergraaf, 2016-10-01:
What exactly is the `import jinja2` problem?

From your log I would guess that there are two python version installed on that machine. The SystemBuilder tools call `python2` which is apparently different from the one that scons uses.

ekiwi, 2016-10-01:
Could it be that OS X isn't following [PEP 394](https://www.python.org/dev/peps/pep-0394/)?
Could you provide us with the output of the following commands:
```
> which python
> which python2
> which python3
```

strongly-typed, 2016-10-01:
Thanks for your quick reply. More investigations showed that Travis CI is f*** up. `scons` starts a different Python version than `python`. The [fix](https://github.com/strongly-typed/xpcc/commit/42bb932b750f1e17db0931c7b140cb0ab66bb5c1) is not to invoke `scons` but `python /usr/bin/scons`. But this is not portable. Any suggestions?

ekiwi, 2016-10-01:
> Any suggestions?

```
> head /usr/bin/scons
> which python
```
and try to find out why Travis has two different python versions installed?

strongly-typed, 2016-10-01:
[CI Environment uses separate virtualenv instances for each Python version. System Python is not used and should not be relied on.](https://docs.travis-ci.com/user/languages/python/#Travis-CI-Uses-Isolated-virtualenvs)

ekiwi, 2016-10-01:
Is it possible to install scons via pip on Travis?

strongly-typed, 2016-10-01:
Good idea! I'll try that!

strongly-typed, 2016-10-02:
Resolved in #193

closed 2016-10-02 by strongly-typed

salkinium, 2016-10-02:
Just FYI: You can also close issues by merging the corresponding PR. See [the docs on this](https://github.com/blog/1506-closing-issues-via-pull-requests).

Thanks!

PR #193 (2016-10-02): [ci] Fix Python dependencies for Travis Trusty.
Close #192