roboterclubaachen/xpcc #205: STM32F7 in non-reset config after DFU mode

strongly-typed, 2016-12-20:
Hi guys, 

while exploring STM32F7 support, we (@daniel-k) discovered that the DFU bootloader leaves the processor in a non-reset configuration, e.g. USART1 is clocked from PCLK2, instead of ABP2. See `RCC->DCKCFGR2`.

This leads to incorrect baud rates when using xpcc.

What is the best place to reset `RCC->DCKCFGR2` to reset defaults? Is it either in the clock config code, or like [xpcc_gpio_enable()](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/gpio/stm32/gpio_enable.cpp.in#L14) in [startup.c](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/startup.c.in)?

Meta:
There may be other configurations which differ from reset-configuration when using the DFU. What assumptions does xpcc make about state of the processor?

Any suggestions appreciated.

salkinium, 2016-12-20:
This would be done [here at the very beginning of the startup script](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/platform.macros#L22). It'll likely be split into two functions in the future, but don't worry about that for now.

> What assumptions does xpcc make about state of the processor?

Power-on reset, which can differ significantly from a debugger reset or when the device resets itself (like DFU). Not sure what to do about that, didn't come up until now.

salkinium, 2016-12-20:
Though technically this is also an issue with the `ClockControl` Implementation that does not correctly configure this bit. This could be solved with #39.

PR #206 (2016-12-20): [platform] Reset clock configuration after DFU bootloader.
Closes #205
closed 2016-12-20 by salkinium