roboterclubaachen/xpcc #211: Error handling in "hosted" code

chris-durand, 2017-01-25:
Hi everyone,

I read some of the hosted specific xpcc code and noticed that there is no consistent error handling strategy. Partially there is no error handling at all. 
Some code just ignores errors and assumes they don't occur, some code prints warnings and continues as if nothing happened, some code can throw exceptions (by using throwing library code), ...

For example, xpcc::hosted::SerialInterface which is used in the CanUSB driver, has a function readBytes that will infinitely loop if the read syscall fails.

Since "Roboterclub Aachen" is planning to run xpcc communication on a raspberry pi in a production environment the lacking error handling of xpcc hosted code is a problem worth to consider.

XPCC is a microcontroller framework, thus it is not at all reasonable to have any error handling for problems that can't occur on a mcu in its basic interfaces.
So we don't have many options. xpcc::Dispatcher for example is a class that is both used on microcontrollers and on hosted. What is the right way to behave for a communication backend on hosted when it encounters an io error?

I can think of three possible behaviours:
- Print an error message and continue to run
- Terminate the program
- Throw an exception and let it propagate to the user of xpcc::Dispatcher. This would require that xpcc::Dispatcher at least fulfills the basic exception safety guarantee. (This could already be true.)

salkinium, 2017-01-25:
> there is no consistent error handling strategy. 

There is now. #185 adds a unified error model to xpcc that works on all platforms (AVR, STM32 and hosted) and does not require use of C++ exceptions. Furthermore it can be used to implement all the behaviors you describe by providing assertion and abandonment handlers. Note that C++ exception cannot be used in xpcc due to them being unavailable on AVRs.

xpcc's error model [builds on research by Duffy](https://gist.github.com/salkinium/66df1d7d3c5d3927690a722a5f880ae0). 
The [assertion interface is documented here](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/interface/assert.hpp).
Also [see](https://github.com/roboterclubaachen/xpcc/blob/develop/examples/avr/assert/main.cpp) [the](https://github.com/roboterclubaachen/xpcc/blob/develop/examples/linux/assert/main.cpp) [examples](https://github.com/roboterclubaachen/xpcc/blob/develop/examples/stm32f469_discovery/assert/main.cpp).

> Partially there is no error handling at all.

True. The assertions are pretty new, and I was too busy with modm to actually make use of them ;-P #114 can be immediately fixed now, which I would actually recommend for a more robust use in our robots.

> What is the right way to behave for a communication backend on hosted when it encounters an io error?

Assert on it, and it will trigger all assertion handlers and, if they vote for abandonment, also the abandonment handler.

salkinium, 2017-01-27:
Did I answer your questions, or did you mean something else, @chris-durand ?

salkinium, 2017-02-02:
ping @chris-durand?

salkinium, 2017-02-15:
Reminder for myself: Add an optional context pointer to the assertion to distinguish between instances at runtime.

salkinium, 2017-03-26:
I'm closing this, since I've improved some aspects of our error model in #217, as well as added a few test uses in #216 and #220.

I've also wrote an [in-depth discussion of the topic on the dev blog](http://blog.salkinium.com/xpccs-error-model).

closed 2017-03-26 by salkinium