roboterclubaachen/xpcc #238: STM32L4: SRAM2 is before SRAM1

strongly-typed, 2017-04-08:
This is easter egg no 1:

On STM32L476 the SRAM1 (`0x2000 0000`) start address is higher than the SRAM2 (`0x1000 0000`) start address:

[image]

In the [Linkerscript template](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/linkerscript/stm32_dccm.ld.in#L166) it is assumed that the order is the other way round.

I even think that `stm32_dccm.ld.in` is the wrong template as the STM32L476 has no Core Coupled Memory (CCM). But I took that from the [feature/stm32l4_support branch](https://github.com/salkinium/xpcc/blob/feature/stm32l476_support/src/xpcc/architecture/platform/driver/core/cortex/driver.xml#L35). (Your second easter egg.)

The resulting linkerscript is not liked by the linker:

     linkerscript.ld:319 cannot move location counter backwards (from 0000000020018000 to 0000000010008000)

I am willing to contribute pull requests for review, separately or in conjunction with a fixed `feature/stm32l4_support` PR?

strongly-typed, 2017-04-08:
I cherry-picked a [new branch ](https://github.com/strongly-typed/xpcc/tree/feature/stm32l4_support_2017q1) based on 2017q1 release for STML4 because it wasn't so easy to rebase the old branch.

salkinium, 2017-04-08:
I think you should add a completely new linkerscript for the L4 series for now, we can then always decide to merge them later, if that's better.

The linkerscript template currently uses the memory name as an identity, which is sub-optimal as shown here. It should really sort the memories by address and use some other mechanism to declare the memory type and use. But I'd prefer to fix that in modm, due to better templating support.

salkinium, 2017-04-08:
Adding all these fixes to `feature/stm32l4_support` is totally fine with me, no need for splitting it up. I see you already have plenty of commits, so it should be easy to review.

strongly-typed, 2017-04-08:
I just leave out the SRAM2 in the current [feature branch](https://github.com/roboterclubaachen/xpcc/pull/240/commits/5cebaaaf99200a3fad2e4a64771b9461341f5214). Until now it's a "works for me".

salkinium, 2017-06-27:
I won't fix this in xpcc anymore, this'll be fixed in modm. This issue remains open though.

salkinium, 2019-01-06:
This has been fixed in modm.

closed 2019-01-06 by salkinium