roboterclubaachen/xpcc #239: STM32L4: Device file wrong

strongly-typed, 2017-04-08:
In the STM32L4 [device file](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/devices/stm32/stm32l476_486-j_m_q_r_v_z-c_e_g.xml#L196) has some issues with the `BreakIn`.
```xml
      <gpio port="A" id="6">
        <af id="1" peripheral="Timer1" name="BreakIn" type="in"/>
        <af id="12" peripheral="Timer1" name="BreakIn" type="in"/>
      </gpio>
```

The generated `gpio.hpp` won't compile because the connect method is overloaded with the same `TypeId`. Your third easter egg ;-)

There are many other occurrences of that bug in that file.

There seems to be an aliasing between different signals to the same name in the parsing of the device files.

[image]

[image]

@salkinium Probably that can only be fixed by you.

salkinium, 2017-04-08:
AF12 should be named `BreakInComp2`. Unfortunately the design of `connect` is broken, because the data doesn't mold itself to our naming abstraction. The place to hotfix that [in the DFG would be here](https://github.com/roboterclubaachen/xpcc/blob/develop/tools/device_file_generator/dfg/stm32/stm_reader.py#L377-L378).

I don't see this truly being fixed in xpcc though. I would recommend removing connect completely and making the `setAlternateFunction()` method public as well as implement a basic pinmux HAL for the F1/L1 series, similarly to how the `SystemClock` is backed by `xpcc::stm32::Clock`.

strongly-typed, 2017-04-08:
I finally [fixed it manually in the device file](https://github.com/roboterclubaachen/xpcc/pull/240/commits/bf65099381b9c4c09f8325fcd1c32b54dd47075a). @salkinium Please take care not to overwrite it when updating the device files.

salkinium, 2017-05-13:
This was fixed in #240.

closed 2017-05-13 by salkinium