roboterclubaachen/xpcc #245: STM32F072 Pwm broken

chris-durand, 2017-04-25:
PWM timers on STM32F072 are broken on some Gpio pins. The following example works perfectly for Output C9 on STMF072-Discovery, but generates no Pwm output when I use Output B1:

```cpp
#include <xpcc/architecture/platform.hpp>

using namespace xpcc::stm32;

// ----------------------------------------------------------------------------
int
main()
{
	Board::initialize();
	
	GpioOutputC9::connect(Timer3::Channel4);
	
	// set up the timer for 16bit PWM
	Timer3::enable();
	Timer3::setMode(Timer3::Mode::UpCounter);

	Timer3::setPrescaler(1);
	Timer3::setOverflow(65535);
	// configure the output channels
	Timer3::configureOutputChannel(4, Timer3::OutputCompareMode::Pwm, 0);

	Timer3::applyAndReset();
	// start the timer
	Timer3::start();
	Timer3::setCompareValue(4, 20000);
	
	while (1)
	{
	}

	return 0;
}
```

This could be related to alternate function remapping.

@salkinium: We would really appreciate to have PWM working on STM32F072 at the German Eurobot competition this weekend ;-)

salkinium, 2017-04-25:
Oh., this weekend? I'm in Aachen from tomorrow for a week. How did I miss that? Damn

salkinium, 2017-04-26:
I'm flying over immediately for a VIP debug session. I'm not kidding.

chris-durand, 2017-04-26:
We will drive to Dresden on Friday morning.

salkinium, 2017-04-26:
And I'm coming with you!

salkinium, 2017-04-26:
ZOMG. I was never more excited about an xpcc issue! LOL.

chris-durand, 2017-04-27:
PEBCAK, hardware issue

closed 2017-04-27 by chris-durand

salkinium, 2017-04-27:
Details: The pins PA2, PA3, PA6, PA7, PB0 and PB1 are not connected to the pinouts by default. To enable this we had to close the solderbridges SB27-SB32. However be aware of the routing of these pins on the PCB to the capacitive touch sensors, which include resistors and capacitors, and may influence the waveform on these pins.