roboterclubaachen/xpcc #252: (Serious?) warnings in unittests for ATmega

strongly-typed, 2017-05-10:
Running

     scons unittest target=atmega

results in

     Warn: Specifying 'general.name' is deprecated and replaced by 'build.name'!

     xpcc/communication/sab/test/slave_test.cpp: In member function 'virtual void SlaveTest::tearDown()':
     xpcc/communication/sab/test/slave_test.cpp:123:9: warning: deleting object of polymorphic class type      'xpcc::sab::Slave<xpcc::sab::Interface<FakeIODevice> >' which has non-virtual destructor might cause      undefined behaviour [-Wdelete-non-virtual-dtor]
       delete slave;
              ^
     xpcc/communication/xpcc/test/dispatcher_test.cpp: In member function 'virtual void      DispatcherTest::tearDown()':
     xpcc/communication/xpcc/test/dispatcher_test.cpp:55:9: warning: deleting object of polymorphic class      type 'TestingComponent2' which has non-virtual destructor might cause undefined behaviour      [-Wdelete-non-virtual-dtor]
       delete component2;
              ^
     xpcc/communication/xpcc/test/dispatcher_test.cpp:56:9: warning: deleting object of polymorphic class      type 'TestingComponent1' which has non-virtual destructor might cause undefined behaviour      [-Wdelete-non-virtual-dtor]
       delete component1;
              ^
     xpcc/communication/xpcc/test/dispatcher_test.cpp:59:9: warning: deleting object of polymorphic class      type 'FakePostman' which has non-virtual destructor might cause undefined behaviour      [-Wdelete-non-virtual-dtor]
       delete postman;
              ^
     In file included from ./unittest/testsuite.hpp:34:0,
                      from xpcc/communication/xpcc/test/dispatcher_test.hpp:34,
                      from xpcc/communication/xpcc/test/dispatcher_test.cpp:32:
     ./unittest/harness.hpp: In instantiation of 'bool unittest::checkEqual(const A&, const B&, unsigned      int) [with A = unsigned int; B = int]':
     xpcc/communication/xpcc/test/dispatcher_test.cpp:169:2:   required from here
     ./unittest/harness.hpp:135:9: warning: comparison between signed and unsigned integer expressions      [-Wsign-compare]
        if (a == b) {
              ^

     In file included from ./unittest/testsuite.hpp:34:0,
                      from xpcc/io/test/io_stream_test.hpp:10,
                      from xpcc/io/test/io_stream_test.cpp:10:
     ./unittest/harness.hpp: In instantiation of 'bool unittest::checkEqual(const A&, const B&, unsigned      int) [with A = int; B = unsigned int]':
     xpcc/io/test/io_stream_test.cpp:559:2:   required from here
     ./unittest/harness.hpp:135:9: warning: comparison between signed and unsigned integer expressions      [-Wsign-compare]
        if (a == b) {
              ^
     In file included from ./xpcc/driver/pressure/bme280_data.hpp:236:0,
                      from ./xpcc/driver/pressure/bme280.hpp:18,
                      from xpcc/driver/pressure/test/bme280_test.cpp:1:
     ./xpcc/driver/pressure/bme280_data_impl_fp.hpp: In member function 'void      xpcc::bme280data::Data::calculateCalibratedTemperature()':
     ./xpcc/driver/pressure/bme280_data_impl_fp.hpp:31:28: warning: left shift count >= width of type
       int32_t adc = ((raw[3] << 16) | (raw[4] << 8) | (raw[5] << 0));
                                 ^
     ./xpcc/driver/pressure/bme280_data_impl_fp.hpp: In member function 'void      xpcc::bme280data::Data::calculateCalibratedPressure()':
     ./xpcc/driver/pressure/bme280_data_impl_fp.hpp:64:28: warning: left shift count >= width of type
       int32_t adc = ((raw[0] << 16) | (raw[1] << 8) | (raw[2] << 0));
                                 ^
     In file included from ./xpcc/driver/pressure/bme280_data.hpp:237:0,
                      from ./xpcc/driver/pressure/bme280.hpp:18,
                      from xpcc/driver/pressure/test/bme280_test.cpp:1:
     ./xpcc/driver/pressure/bme280_data_impl_double.hpp: In member function 'void      xpcc::bme280data::DataDouble::calculateCalibratedTemperature()':
     ./xpcc/driver/pressure/bme280_data_impl_double.hpp:33:28: warning: left shift count >= width of type
       int32_t adc = ((raw[3] << 16) | (raw[4] << 8) | (raw[5] << 0));
                                 ^
     ./xpcc/driver/pressure/bme280_data_impl_double.hpp: In member function 'void      xpcc::bme280data::DataDouble::calculateCalibratedPressure()':
     ./xpcc/driver/pressure/bme280_data_impl_double.hpp:65:28: warning: left shift count >= width of type
       int32_t adc = ((raw[0] << 16) | (raw[1] << 8) | (raw[2] << 0));
                                 ^
     Compiling C++: /Users/usr/Dev/2017amt/xpcc_feature_rtc_ds1302_support/build/unittest_atmega/xpcc/math/     filter/test/debounce_test.o
     xpcc/driver/pressure/test/bmp085_test.cpp: In member function 'void Bmp085Test::testConversion()':
     xpcc/driver/pressure/test/bmp085_test.cpp:117:14: warning: comparison between signed and unsigned      integer expressions [-Wsign-compare]
         if (error > max_error) {
                   ^
     xpcc/driver/pressure/test/bmp085_test.cpp:182:15: warning: comparison between signed and unsigned      integer expressions [-Wsign-compare]
          if (error > max_error) {
                    ^

The current unittests can only by run on an ATMega2560. Others do not have enough flash.

     Device: atmega2560

     Program:  242610 bytes (92.5% Full)
     (.text + .data + .bootloader)

I will try to work on the BME/BMP issues. What about the other warnings?

strongly-typed, 2017-05-11:
Fixed the shift and signedness warnings [here](https://github.com/strongly-typed/xpcc/tree/fix/release_tests). What about the non-virtual destructor?

salkinium, 2017-05-11:
> What about the non-virtual destructor?

Empty virtual destructors would suffice, I think.

PR #253 (2017-05-11): Fix shift errors and warnings in release tests
This closes #252.
closed 2017-05-11 by salkinium