roboterclubaachen/xpcc #257: Linker error due to global variable included by display.h

cajt, 2017-06-14:
in src/xpcc/driver/display/siemens_m55_impl.hpp: "initData_lm15[]" 
This global variable can cause a linker error if display.h is included from multiple compilation units, as this will create redundant entries in .data section.

Output
Linking: /home/user/development/test/build/main.elf
/home/user/development/test/src/app.o:(.data.initData_lm15+0x0): multiple definition of `initData_lm15'
/home/user/development/test/src/main.o:(.data.initData_lm15+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/4.8/../../../arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions

salkinium, 2017-06-14:
Yes, that's just wrong. Keep in mind that this display driver is very old and wasn't ported to the new SpiMaster interface and will therefore not work even if compilation succeeds.

salkinium, 2017-06-28:
Fixed in https://github.com/roboterclubaachen/xpcc/commit/611e3afbbc92f0bd01cf104de7f1fa447de75f40.

closed 2017-06-28 by salkinium