roboterclubaachen/xpcc #264: GDB peripheral register view

AndreGilerson, 2017-07-04:
The current version of xpcc does not include symbols for the various peripheral registers, making it hard to debug applications, since one has to lookup the address of a register in the datasheet. 

[This project](https://github.com/maximevince/ucregview) adds an additional .elf file overwriting the defines from the STM header with additional symbols pointing to the correct addresses. This allows for easy debugging by looking at peripheral registers via the e.g. `p/x USART1` command.

Does anybody know a different solution, or should this be added to xpcc? Can we generate a file similar to the aforementioned project from the device files? Otherwise one could write a script parsing the corresponding header.

salkinium, 2017-07-04:
> Does anybody know a different solution, or should this be added to xpcc?

Yes please, this is a very wonderfully useful idea!

> Can we generate a file similar to the aforementioned project from the device files? 

Nope. Device files do not contain a memory map (and I can't easily add that data for now anyways). In theory the [CMSIS-SVD files](https://github.com/posborne/cmsis-svd) contain the memory map, but their quality and device coverage is very poor, I strongly suggest you do not use them.

> Otherwise one could write a script parsing the corresponding header.

That would be the best approach. It also has the advantage of always being in sync, since you're using the same source files for both compiling C and generating the elf.

salkinium, 2017-07-06:
This is such a great idea I, uh, ~accidentally~ implemented this for modm already (theoretically for all STM32, currently confirming that) and it works pretty great. See [the device header parser and symbol resolver in this commit](https://github.com/modm-io/modm/commit/45ed70a1ced1d07611960a34b5461ab1e71bffb6#diff-a2fed1aded63972e79d485660e8c7dcaR76).
For my current test platform STM32F469 [this generates these two files](https://gist.github.com/salkinium/3729ba34fd2192685c8c332009b90f15).
As you can see the linkerscript is way simpler than in the project, since you can just assign any objects address directly (`EXTERN` is required to preserve the type, otherwise type `word` is assumed by linker):
```
EXTERN(TIM2);
TIM2 = 0x40000000;
EXTERN(TIM3);
TIM3 = 0x40000400;
```
The main linkerscript then simply `INCLUDE peripherals.ld` and BAM, debuggable peripherals. Keep in mind though that reading some register can have side-effects, so be careful what you read.

This is missing the generation for the ARM Cortex-M peripherals, I'm adding that right now.

AndreGilerson, 2017-07-07:
I looked into the [commit](https://github.com/modm-io/modm/commit/45ed70a1ced1d07611960a34b5461ab1e71bffb6#diff-a2fed1aded63972e79d485660e8c7dcaR76) but was not able to get the script working on my machine (Ubuntu 16.04, Python 3.5). The Problem was in line 86:
`for symbol, value in {m[0]: m[1] for m in rest}.items():`
The defines we are looking for a either absolute, containing an address, or relative, depending on an absolute define and adding an offset. In the header file the relative defines always come after the absolute ones, since they depend on them. In line 86 you create a dictionary out of those defines: However, python does not guarantee that the dictionary keeps the order of its entries. This breaks the script for me, as sometimes a relative define is processed before the necessary absolute one was put into the `symbols` dictionary, failing to resolve the address. This can easily be fixed by forcing the dictionary to keep the order of its entries. Add an `from collections import OrderedDict` and change line 86 to:
`for symbol, value in 'OrderedDict((m[0]: m[1] for m in rest).items():`

Furthermore a second problem arose, when i was trying to use the script with the `stm32l476xx.h` header. A slightly different format was used in that file, by explicitly adding a (uint32_t) to the addresses definitions, e.g:
`#define FLASH_BASE     ((uint32_t)0x08000000U)`
I therefore suggest changing the regular expression in line 84 to:
`rest = re.findall(r"#define +(.*?) +\(?(?:\(uint32_t\))?(.*?)U?\)? *?(\n|/)", memmapContent)`
This allows it to throw out the uint32_t statement, allowing the same script to also process the `stm32l476xx.h` header.

salkinium, 2017-07-07:
Thanks! I've [updated the script](https://github.com/modm-io/modm/commit/13a00cb27f591cea0ec216259d214e27dac225d8) to also include the Cortex-M peripherals and tested it on 801 STM32 devices. I had to [fix some STM32 header files](https://github.com/modm-io/cmsis-header-stm32/commit/3ed38e94f93eb4ac3527a9936160e4db1e4eced2), I sent a bug report to ST.
[image]

salkinium, 2017-07-09:
Just to communicate this clearly: I don't intend on back-porting this to xpcc myself, due to lack of motivation. I think @strongly-typed knows enough about the build system to be able to add it though and I would be happy to merge that (even if it's an ugly hack).

daniel-k, 2017-07-24:
Maybe I'm missing something here, but doesn't adding `-g3` to the compile commandline yield the same benefit?

```
>>> p/x *SysTick
$6 = {
  CTRL = 0x10007, 
  LOAD = 0x2903f, 
  VAL = 0x196b8, 
  CALIB = 0x4000493e
}
```

salkinium, 2017-07-24:
> Level 3 includes extra information, such as all the macro definitions present in the program. Some debuggers support macro expansion when you use -g3.

Cool! Yeah, maybe that is already enough. xpcc build system needs a release and debug build mode though.

daniel-k, 2017-07-24:
`-g3` doesn't hurt any performance to my knowledge since it resides in the binary on the host and none of the additional information is flashed to the target. If that's what you're implying ...

salkinium, 2017-12-18:
Fixed with `-g3` in #266.

closed 2017-12-18 by salkinium