roboterclubaachen/xpcc #265: [avr] Atomic queue fails on ATmega2560

strongly-typed, 2017-07-06:
While trying to set up unittests on AVR again I was experimenting with an ATmega2560 on a Chinese Mega 2560 clone. I am programming the board directly by ISP with a USBASP clone. This seems to work fine.

I can run a blinky on the board. As soon as I add `uart::initialize<clock, 9600>();` to the source code the board crashes. I narrowed it down that the usage of `static xpcc::atomic::Queue<uint8_t, 8> rxBuffer` kills the board.

Having a working blinky:

    #include <xpcc/architecture/platform.hpp>
    #include <xpcc/architecture/driver/atomic/queue.hpp>

    using namespace xpcc::atmega;
    typedef xpcc::avr::SystemClock clock;

    typedef GpioOutputB7 Led;
    static xpcc::atomic::Queue<uint8_t, 8> rxBuffer;

    int
    main()
    {
        Led::setOutput();

        while(true)
        {
            // if (rxBuffer.isEmpty()) {
                xpcc::delayMilliseconds(100);
            // }

            Led::toggle();
            xpcc::delayMilliseconds(100);
        }
    }

As soon as I remove the comments the board does not blink anymore.

I am using

     Using built-in specs.
     Reading specs from /usr/local/Cellar/avr-gcc/6.3.0/lib/gcc/avr/6.3.0/device-specs/specs-avr2
     COLLECT_GCC=avr-g++
     COLLECT_LTO_WRAPPER=/usr/local/Cellar/avr-gcc/6.3.0/libexec/gcc/avr/6.3.0/lto-wrapper
     Target: avr
     Configured with: ../configure --target=avr --prefix=/usr/local/Cellar/avr-gcc/6.3.0      --enable-languages=c,c++ --with-gnu-as --with-gnu-ld --with-ld=/usr/local/opt/avr-binutils/bin/avr-ld      --with-as=/usr/local/opt/avr-binutils/bin/avr-as --disable-nls --disable-shared --disable-threads      --disable-libssp --disable-libstdcxx-pch --disable-libgomp --with-gmp=/usr/local/opt/gmp --with-mpfr=/usr/     local/opt/mpfr --with-mpc=/usr/local/opt/libmpc --with-system-zlib
     Thread model: single
     gcc version 6.3.0 (GCC)     

This can be a strange issue with linker scripts or what so ever. Any ideas?

ekiwi, 2017-07-06:
> ATmega2560 on a Chinese Mega 2560 clone.

Is the chip from ATMEL or also a Chinese clone?

salkinium, 2017-07-06:
> Any ideas?

Does behavior change if you enable/disable interrupts manually? On AVRs you need to enable interrupts manually, on Cortex-M they are already enabled (maybe we want to change that?).

salkinium, 2017-07-06:
It could be that avr-gcc 6.3.0 generates the wrong assembly for this function. Keep in mind that Atmel does not upstream all their changes to gcc, so maybe check the assembly.

strongly-typed, 2017-07-07:
Thank you for you responses.

I can not assure that it is a genuine chip from Atmel.

Adding `sei();` does not change the behaviour.

Sigh, I really have to look into the assembly. Compiler is from `osx-cross/avr/avr-gcc`. Does our TravisCI provides artifacts I can use?

salkinium, 2017-07-07:
> Does our TravisCI provides artifacts I can use?

Yes, but [for linux only](http://box.xpcc.io). I made a [homebrew tap for the official avr-gcc toolchain](https://github.com/salkinium/homebrew-atmel-avr) from Atmel. Unfortunately it clashes with arm-none-eabi-gcc, so I'm still using the `osx-cross` avr-gcc 6.2.0 (6.3.0 also clashes }:-( )

strongly-typed, 2017-07-10:
OK, lessons learned: Know your tools. 

The underlying problem is in the V1.02 firmware version of the Chinese "USBASP V2.0" clone. I don't want to blame the Chinese manufacturer. The bug seems to be in the original firmware. It does not work properly with devices with more than 128 KiByte flash. [The Internet knows the bug and how to fix it.](https://petervanhoyweghen.wordpress.com/2015/12/02/the-usbasp-and-atmega2560-mystery/)

How to reproduce the problem: Use USBASP with firmware < 1.05 and do

     dd if=/dev/urandom bs=4096 count=64 of=rnd.bin
     avrdude -p atmega2560 -c usbasp -U flash:w:rnd.bin:r

The result should be like "262144 bytes of flash verified", but it is not with V1.02. Firmware V1.02 only works up to 131072 bytes and leaves a slightly corrupted firmware in the flash on ATMega2560. I really can't tell why it worked without the constructor and can't tell why it did not work with the constructor even though less than 128 KiByte flash was used.

Using V1.05 firmware works flawlessly on two boards.

I close the issue and reopen it if necessary.

Btw, it was not the [global constructor bug in avr-gcc](http://andybrown.me.uk/2010/10/24/the-major-global-constructor-bug-in-avr-gcc/).

closed 2017-07-10 by strongly-typed

salkinium, 2017-07-10:
> The underlying problem is in the V1.02 firmware version of the Chinese "USBASP V2.0" clone. 

Good to know, I bought one of those as well. Haven't used it yet though.