roboterclubaachen/xpcc #268: [stm32f1] Support for STM32F107 is broken

strongly-typed, 2017-07-19:
Enabling the disabled STM32F107 in the [SConscript.check](https://github.com/roboterclubaachen/xpcc/blob/develop/SConscript.check#L101) fails.

```
In file included from /Users/usr/Dev/2017amt/xpcc_feature_stm32f107/build/release/test/stm32/libxpcc/src/xpcc/architecture/../../../generated_platform/drivers.hpp:112:0,
                 from /Users/usr/Dev/2017amt/xpcc_feature_stm32f107/build/release/test/stm32/libxpcc/src/xpcc/architecture/platform.hpp:19,
                 from /Users/usr/Dev/2017amt/xpcc_feature_stm32f107/src/xpcc/architecture.hpp:41,
                 from main.cpp:2:
/Users/usr/Dev/2017amt/xpcc_feature_stm32f107/build/release/test/stm32/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/gpio/stm32f1/gpio.hpp: In static member function 'static void xpcc::stm32::GpioOutputA11::connect(xpcc::stm32::TypeId::UsbDm)':
/Users/usr/Dev/2017amt/xpcc_feature_stm32f107/build/release/test/stm32/libxpcc/src/xpcc/architecture/../../../generated_platform/driver/gpio/stm32f1/gpio.hpp:6644:33: error: expected primary-expression before '<<' token
   AFIO->MAPR = (AFIO->MAPR & ~( << 10)) | ( << 10);
                                 ^~
```

This check has been deactivated [some time ago](https://github.com/roboterclubaachen/xpcc/blame/develop/SConscript.check#L97). Not sure if it ever worked.

Seems to be another device file issue?

@salkinium Could you please confirm?

salkinium, 2017-07-19:
Yes, the offending [device data are the USB ids](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/devices/stm32/stm32f105_107-r_v-8_b_c.xml#L175):
```xml
<af id="10" peripheral="Usb" name="Dm"/>
<af id="10" peripheral="Usb" name="Dp"/>
```

salkinium, 2019-01-06:
This has been fixed in modm.

closed 2019-01-06 by salkinium