roboterclubaachen/xpcc #286: Multiple non-critical warnings (Scons & Jinja2)

cajt, 2017-08-28:
Scons outputs multiple warning for some targets: "scons: warning: Two different environments were specified for target "(..).

This happens for example with: example/nucleo_f411re/blink and nucleo_f103rb/blink/
but not with: nucleo_f401re/blink or stm32f4_discovery/blink .

commit: 195af9aaccb0e66e1c29cbab688d9e55b1e7ef2b

How to reproduce:

```
git clone https://github.com/roboterclubaachen/xpcc.git
cd xpcc/examples/nucleo_f103rb/blink/
scons
```

And this what SCons ouptuts (shortened in the middle):

```

user@box:~/projects/xpcc/examples/nucleo_f103rb/blink$ scons
scons: Reading SConscript files ...

scons: warning: Two different environments were specified for target /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/spi/stm32/spi_base.hpp,
	but they appear to have the same action: jinja2_template_action(target, source, env)
File "/home/user/projects/xpcc/scons/site_tools/platform_tools.py", line 195, in platform_tools_generate

scons: warning: Two different environments were specified for target /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/spi/stm32/type_ids.hpp,
	but they appear to have the same action: jinja2_template_action(target, source, env)
File "/home/user/projects/xpcc/scons/site_tools/platform_tools.py", line 195, in platform_tools_generate

scons: warning: Two different environments were specified for target /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/spi/stm32_uart/type_ids.hpp,
	but they appear to have the same action: jinja2_template_action(target, source, env)
File "/home/user/projects/xpcc/scons/site_tools/platform_tools.py", line 195, in platform_tools_generate

scons: warning: Two different environments were specified for target /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/uart/stm32/uart_base.hpp,
	but they appear to have the same action: jinja2_template_action(target, source, env)
File "/home/user/projects/xpcc/scons/site_tools/platform_tools.py", line 195, in platform_tools_generate

scons: warning: Two different environments were specified for target /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/uart/stm32/type_ids.hpp,
	but they appear to have the same action: jinja2_template_action(target, source, env)
File "/home/user/projects/xpcc/scons/site_tools/platform_tools.py", line 195, in platform_tools_generate

scons: warning: Two different environments were specified for target /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/uart/stm32/uart_baudrate.hpp,
	but they appear to have the same action: jinja2_template_action(target, source, env)
File "/home/user/projects/xpcc/scons/site_tools/platform_tools.py", line 195, in platform_tools_generate
scons: done reading SConscript files.
scons: Building targets ...
Template: '/home/user/projects/xpcc/src/xpcc/architecture/platform/platform.hpp.in' to '/home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/src/xpcc/architecture/platform.hpp'
Template: '/home/user/projects/xpcc/src/xpcc/architecture/platform/drivers.hpp.in' to '/home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/drivers.hpp'
Template: '/home/user/projects/xpcc/src/xpcc/architecture/platform/driver/adc/stm32f1/type_ids.hpp.in' to '/home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/adc/stm32f1/type_ids.hpp'

(...)

Compiling C++: /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/build/nucleo_f103rb/blink/libxpcc/generated_platform/driver/gpio/stm32f1/gpio_enable.o
Compiling C: /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/ext/tlsf/tlsf.o
Create Library: /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/src/libxpcc.a
Indexing: /home/user/projects/xpcc/build/nucleo_f103rb/blink/libxpcc/src/libxpcc.a
Linking: /home/user/projects/xpcc/build/nucleo_f103rb/blink/blink.elf
Creating load file for Flash: /home/user/projects/xpcc/build/nucleo_f103rb/blink/blink.hex
Size after:
Memory Usage
------------
Device: stm32f103rb

Program:    3692 bytes (2.8% used)
(.data + .fastcode + .fastdata + .reset + .rodata + .text)

Data:       5404 bytes (26.4% used) = 2332 bytes static (11.4%) + 3072 bytes stack (15.0%)
(.bss + .data + .fastcode + .fastdata + .noinit + .stack + .vectors)

Heap:      15160 bytes (74.0% available)
(.heap1)

scons: done building targets.

```

ekiwi, 2017-08-28:
This is a known bug, where we generate two identical build descriptions for a single target. We never found the time to debug and have been ignoring it for the last couple of years.

Thanks for reporting it, so that we have a clear description in our issue tracker!

salkinium, 2017-08-29:
MODM will fix _all_ the problems. Just… a tiny bit more yak shaving, then it's ready. 😇

salkinium, 2019-01-06:
This has been fixed in modm.

closed 2019-01-06 by salkinium