roboterclubaachen/xpcc #287: LD segfaults when compiling the nucleo_f031k6 blinky example

riktw, 2017-08-31:
Hello,

I tried compiling the nucleo_f031k6 blinky example and I got the following error:
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
compilation terminated.

When I compile a different example, the nucleo_f103rb or stm32f072_discovery it works as expected.

I am running Debian 9 as OS and I tried it with both the 2017 q2 release and the latest version from git.
It seems that there is something wrong with the nucleo_f031k6 example or configs as other examples compile just fine.
Any idea what it could be?

salkinium, 2017-08-31:
GNU Linker is buggy as heck for Cortex-M.
We've had [a similar issue before](https://github.com/roboterclubaachen/xpcc/issues/159), which eventually went away when we upgraded toolchains.
There's nothing you can do, except trial-and-error: poking around the linkerscript or moving objects around files until it doesn't crash anymore. I've tried to get a minimal example with this behavior, but I failed which was quite frustrating.

salkinium, 2017-08-31:
So the linker doesn't crash in [our Travis CI environment running Ubuntu 14.10](https://travis-ci.org/roboterclubaachen/xpcc/jobs/268429797#L1193) and also not on my local machine (OS X 10.12) both with the 2017Q2 toolchain.

We are using the [official pre-compiled toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) though, maybe Debian distributes their own compiled version?

riktw, 2017-08-31:
Hey

Thanks, I installed the latest (Version 6-2017-q2-update) version of the toolchain from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads instead of the one installed by using apt install and now it works as expected :)

closed 2017-08-31 by riktw