roboterclubaachen/xpcc #289: ST Cube outdated

strongly-typed, 2017-09-04:
ST Cube was updated.

     ➜  st git:(develop) python3 check_for_updates.py
     f0: v1.7.0 -> v1.8.0	update!
     f1: v1.4.0 -> v1.6.0	update!
     f2: v1.4.0 -> v1.6.0	update!
     f3: v1.7.0 -> v1.9.0	update!
     f4: v1.15.0 -> v1.16.0	update!
     f7: v1.6.0 -> v1.7.0	update!
     l0: v1.8.0 -> v1.9.0	update!
     l1: v1.6.0 -> v1.7.0	update!
     l4: v1.7.0 -> v1.8.0	update!

Can this automatically monitored by a TravisCI job?

salkinium, 2017-09-04:
This _is_ automatically monitored by a TravisCI job (https://travis-ci.org/modm-io/cmsis-header-stm32), but it only runs every 24h, which I think is ok.

salkinium, 2017-09-04:
The subtree needs to be updated. I was planning on only doing that once a release, but maybe that's not acceptable?

strongly-typed, 2017-09-05:
Great to see that job. Did not know about it. 

IMHO once per release is enough, but it should be updated timely before a release so that some testing can take place. ST definitely did not introduce any new bugs `</irony>`.

PR #295 (2017-10-01): Update STM32 CMSIS headers for 2017q3 release
Fixes #289.
closed 2017-10-01 by salkinium