roboterclubaachen/xpcc #291: Investigate alternatives for Travis CI

salkinium, 2017-09-21:
I'm getting very tired of Travis CI, it needs a way too much hand holding for something as simple as compiling our code. It's getting ridiculous.

We need to control the environment ourselves and not let Travis yank the rug out of under our feet like with their [latest backwards-incompatible idiocy](https://github.com/roboterclubaachen/xpcc/pull/290).
I want something where we can upload our own Docker container (@strongly-typed already [has](https://github.com/strongly-typed/docker-xpcc-unittest-stm32) [a](https://github.com/strongly-typed/docker-arm-none-eabi-gcc) [few](https://github.com/strongly-typed/docker-xpcc-build-arm) :-).

salkinium, 2017-09-21:
[CircleCI has first-class support for Docker.](https://circleci.com/features/)

salkinium, 2017-09-21:
[There's also LambCI](https://medium.com/@hichaelmart/lambci-4c3e29d6599b), but it may be a little too advanced for what we need here.

salkinium, 2017-09-23:
Ok, I was wrong, it's not Travis CI faults, SCons has been [updated to 3.0.0](http://scons.org/scons-300-is-available.html) and it has support for Python 3.5+. This is pretty exciting!

cc @strongly-typed @dergraaf @ekiwi @daniel-k

salkinium, 2017-09-23:
Sorry Travis CI, we still love you (somewhat).

salkinium, 2017-09-23:
I hotfixed this issue in #292.

strongly-typed, 2017-10-02:
Yeah, really love Docker (who does not?).
Just added to the maybe-someday list. Probably November+ is a good time to set this up, maybe in conjunction with CI-efforts at RCA ;-)

strongly-typed, 2017-10-11:
So, was too excited about how CircleCI performs, so I just opened #298. Will you merge?

salkinium, 2017-10-11:
I like opening issues and then magically they get implemented. Very nice!

salkinium, 2017-10-15:
Fixed by #298.

closed 2017-10-15 by salkinium