roboterclubaachen/xpcc #303: Cortex hardfault handler only made for STM32

strongly-typed, 2017-10-24:
[`<parameter name="hardfault_handler_led_port" type="enum" values="A;B;C;D;E;F;G;H;I;J;K">A</parameter>`](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/architecture/platform/driver/core/cortex/driver.xml#L17)

is not suitable for LPCs as their port naming scheme is `GPIO0_1` et cetera.

I see that this might be fixed in mdom.

salkinium, 2017-10-24:
Yes, this is fixed in modm by
1. providing the [option enumeration with the exact pinout of the selected device](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/module.lb#L66), and
2. dropping LPC support altogether 😇

strongly-typed, 2017-10-25:
> dropping LPC support altogether

I just sold all my NXP stock 📉

So any reasoning for that? Sure, RCA does not use NXP anymore, but ...

salkinium, 2017-10-25:
I don‘t have any machine readable data for the LPCs to generate the new version of device files, which contain a lot more information. In general, machine readable data is *the* bottleneck for modm at the moment.

strongly-typed, 2017-10-25:
So let's don't say "Drop LPC support" but [keep that single file](https://github.com/modm-io/modm/tree/develop/devices/lpc).

If machine readable data comes available the LPC support can be expanded. Having a second vendor (or at best a third, fourth) is a good benchmark if the concepts from xpcc/modm work beyond the ST-world (see this issue).

salkinium, 2017-10-25:
> So let's don't say "Drop LPC support" but keep that single file.

I didn't just arbitrarily drop LPC support, there was a long process of realization that it won't work without significant manual work. The drivers require certain information to be present in all devices, and I quite frankly wasn't interested in keeping that file up-to-date with my many changes to the device file format and content from the AVR/STM32 generators.

> If machine readable data comes available the LPC support can be expanded. 

Absolutely. But this data isn't available now. I'm dealing with that issue by dropping support.

> Having a second vendor (or at best a third, fourth) is a good benchmark if the concepts from xpcc/modm work beyond the ST-world (see this issue).

Yes, having data on multiple **vendors** is great, but a single device is not much help. We missed the STM32F1 GPIO MUX problems too, even though we had the data.

I'm sorry for dropping LPC support in modm, but both @dergraaf and I have to prioritize our work for both time and motivation. As a compromise I haven't deleted the LPC code in modm, it just won't show up as a target.

salkinium, 2018-02-02:
Note that I've refactored the hardfault handler in modm and [placed it into it's own module](https://github.com/modm-io/modm/commit/4e134ff448ff1982b417aae4797d4da47b409401#diff-8ac86f1b379d79be07856191efc4f883) and [threw out the option to print an error stack via UART](https://github.com/modm-io/modm/commit/6ce957ed313a7e67fb4614dcc891f7478442f39b). We should integrate [CrashCatcher](https://github.com/adamgreen/CrashCatcher) instead.

salkinium, 2019-01-06:
Moved to modm.

closed 2019-01-06 by salkinium