roboterclubaachen/xpcc #308: AVR target is broken?

siy, 2017-11-02:
I've found that compiling for AVR target (arduino uno board) fails with following diagnostics:
`xpcc/src/xpcc/math/geometry/location_2d_impl.hpp:169:51: error: call of overloaded 'abs(float)' is ambiguous
    (std::abs(this->orientation - other.orientation) < __FLT_EPSILON__));`

Reproduction is very simple: clone getting-started-with-xpcc repo, change board in project.cfg to arduino_uno and then run scons.

siy, 2017-11-02:
I've found that this is not specific to arduino_uno board, simple example projects like examples/avr/gpio/(basic/blinking/button_group) are failing as well, with same diagnostics.

chris-durand, 2017-11-02:
I can confirm this problem on Arch linux with avr-g++ 7.2.0. Somehow including `stdlib.h` in graphic_display.cpp breaks overload resolution for std::abs(float). Changing the include to `cstdlib` fixes compilation for AVR.

salkinium, 2017-11-02:
Hm, that's interesting. Our CI isn't complaining and I can't reproduce this locally either with `examples/avr/gpio/blinking` or `getting-started-with-xpcc`. 🤔

What avr-gcc version are you using on what OS? (I'm using `gcc version 5.4.0 (AVR_8_bit_GNU_Toolchain_3.6.0_487)` on macOS, our CI is using v4.9).

salkinium, 2017-11-02:
> Changing the include to `cstdlib` fixes compilation for AVR.

Oh ok, nvm my comment then. Do you want to create a PR?

We're only compiling against the official toolchain from Atmel, but it's a bit out-of-date, see [also this issue on C++14 support](https://github.com/roboterclubaachen/xpcc/issues/29).

siy, 2017-11-02:
I also have 7.2.0.

chris-durand, 2017-11-02:
> Do you want to create a PR?

Done (#309)

PR #309 (2017-11-02): [avr] Fix compilation for avr gcc 7.2.0
Fixes #308
closed 2017-11-02 by salkinium