roboterclubaachen/xpcc #316: [ci

strongly-typed, 2017-11-21:


salkinium, 2017-11-21:
I built them on the Ubuntu server that hosts them, so that TravisCI doesn’t have to build them over and over again. I already forgot how I did that. I’m ready to remove Travis CI and only use Circle CI.

salkinium, 2017-11-21:
Would also break when I remove that server or misconfigure the subdomain or something, so we should definitely not rely on it.

strongly-typed, 2017-11-21:
In TravisCI, we use the compiler you provide. Is seems to be a compiler from Atmel.
`gcc version 5.4.0 (AVR_8_bit_GNU_Toolchain_3.4.4_201706301059)`. The `3.4.4` is definitely wrong. This is caused by a bug in Atmel's build script. Hey Atmel, do you have QA?

Why exactly don't we use the `avr-gcc` we could install in TravisCI's Ubuntu? Because Travis is based on Ubuntu 14.04 and the compiler in 14.04 is too old? Travis seems to stick to good old 14.04.

In CircleCI, we use the [AVR GCC from Ubuntu 16.04](https://github.com/strongly-typed/docker-xpcc-build/blob/master/Dockerfile#L24), which is not a compiler from Atmel.

salkinium, 2017-11-21:
> In CircleCI, we use the AVR GCC from Ubuntu 16.04, which is not a compiler from Atmel.

My hope was that by using the canonical compiler from Atmel we could avoid issues with the different custom compiled avr-gcc packages by different Linux distributions. However, that requires the end user to actually install the (outdated) Atmel avr-gcc. To my limited experience with our linux devs, that's not what happens, they just install whatever their distributions package manager installs, and so we should also just use the latest avr-gcc from upstream.

The question is really about device coverage, because the upstream avr-gcc fails to compile/assemble for more AVR devices than the Atmel avr-gcc (I assume they don't upstream their changes). 
I'd be ok with using the latest upstream avr-gcc, either precompiled from the distribution or shipped as a Docker container.

To be honest, I don't really care that much about AVRs anymore, since Cortex-M is just waaaay less of a hassle in the long run. Just thinking about the hacks required to access constant data in flash on AVRs makes me cringe.

strongly-typed, 2017-11-22:
Yes, I don't care much about AVRs as well. Their technology is limited and cumbersome sometimes. The only benefit is cheap and widespread Arduino hardware. But the blue pill and Nucleos are becoming a better alternative.

We could use [Github releases](https://developer.github.com/v3/repos/releases/#upload-a-release-asset) to provide the tar ball [here](https://github.com/strongly-typed/build-atmel-avr-gnu-toolchain/releases).

CircleCI successfully built the [artifact](https://circleci.com/gh/strongly-typed/build-atmel-avr-gnu-toolchain/7#artifacts/containers/0) yesterday for me. The last step would be to publish this artifact to Github automatically. Then, TravisCI (and any user) could rely on this toolchain.

I have to admit that this is more about demonstrating methods on how to streamline such a process. I really don't care much about AVRs.

strongly-typed, 2017-11-24:
So, this now [seems to work](https://travis-ci.org/strongly-typed/xpcc/jobs/306661742#L964): TravisCI now uses the latest Atmel toolchain compiled on CircleCI.

What about the boost.tar.bz2?

strongly-typed, 2017-12-18:
> What about the boost.tar.bz2?

Friendly ping.

salkinium, 2017-12-18:
> What about the boost.tar.bz2?

Also built and hosted on my private server, but didn't document how. I think it would be good to create the shell script `/tools/build_boost.sh` to automatically build this from source, then host it on GitHub as a resource. 

Maybe boost should be replaced with the C++ standard library? It is mostly used for [things like threading and IO](https://github.com/roboterclubaachen/xpcc/search?utf8=✓&q=boost&type=) which should by now be implemented natively in the stdlib++.

salkinium, 2018-01-07:
The only dependence we now have on hosted is Boost.Asio, which exists also as a standalone version.
https://github.com/roboterclubaachen/xpcc/pull/322#issuecomment-353509746

PR #330 (2018-01-13): [ci] Remove TravisCI in favour of CircleCI
I've had enough of TravisCI, it's both slower and worse to manage than CircleCI. We're already running on a deprecated version and it's just silly to have to download precompiled packages from a personal server or cache on every boot.

cc @strongly-typed This closes #316.
closed 2018-01-13 by salkinium