roboterclubaachen/xpcc #317: Support for half-duplex SPI

genbattle, 2017-12-11:
The STM32 MCUs support half-duplex SPI, which is needed to talk to some peripherals like the [LSM303](http://www.st.com/en/mems-and-sensors/lsm303c.html) magnetometer/accelerometer. It would be great if this were exposed via XPCC's SPI API.

salkinium, 2017-12-12:
I think the data API still works for Half-Duplex by sending dummy bytes (as done in several drivers already). So we need to add a function to the SpiMaster API to enable half-duplex mode in hardware, right?

genbattle, 2017-12-12:
That sounds right. Possibly you could automatically enable half-duplex mode when the MOSI and MISO pins are the same and keep the same API?

salkinium, 2017-12-12:
Hah. I wrote a driver for the [LSM303 accelerometer](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/driver/inertial/lsm303a.hpp#L370) on the [F3 Discovery](https://github.com/roboterclubaachen/xpcc/blob/develop/examples/stm32f3_discovery/accelerometer/main.cpp) some time ago. I knew it sounded familiar to me. This is a bit of a weird chip, it almost seems like ST just put two dies in there and connected their busses internally, but not quite.

These sensors all use a similar dual I2C/SPI interface, that is shared with a lot of other ST sensors.
This is abstracted in [the `Lis3Transport` class](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/driver/inertial/lis3_transport.hpp), which *drumroll* always uses [half-duplex transmissions in SPI mode](https://github.com/roboterclubaachen/xpcc/blob/develop/src/xpcc/driver/inertial/lis3_transport_impl.hpp#L106) by sending dummy-bytes on read and ignoring read bytes on write.

The dummy bytes are signalled with a `nullptr` I think because the value was platform dependent? Not sure, but it's abstracted already. I think this should just automagically work by adding a function to the SpiMaster to enable this hardware mode.

(I remember the magnetometer being very similar to some other magnetometer sensor driver in xpcc, which is why I didn't write a driver for it and then forgot…)

genbattle, 2017-12-12:
Wow now I really feel like an idiot, I've literally been writing my own drivers for the LSM303 and LIS3MDL for the last couple of months. I can't believe I missed that xpcc already has a driver for these devices.

Ok, so given that the only work that needs to be done is adding support for the hardware mode, what's the best way to do this? I like the idea of automatically enabling it if the MOSI and MISO pins are the same since it doesn't require modifying the existing interface at all, and it doesn't leave a useless method on other platforms. The disadvantage would be that it's somewhat unintuitive without clear documentation of this feature.

If all that is needed is adding the hardware support, this could be something I could get done over the Christmas holidays.

chris-durand, 2017-12-13:
> These sensors all use a similar dual I2C/SPI interface, that is shared with a lot of other ST sensors.
> This is abstracted in the Lis3Transport class, which drumroll always uses half-duplex transmissions in SPI mode by sending dummy-bytes on read and ignoring read bytes on write.

I suppose this is not what @genbattle means. The LSM303 also supports a half-duplex 3-wire SPI mode which uses only one bidirectional data line (no MISO/MOSI; only SCK, CS and SDIO). The STM32 has hardware support for this. You have to set the BIDIMODE bit in SPI_CR1 to enable this mode. This will disable the MISO pin. The MOSI pin will be used as a bidirectional IO. The data direction can be controlled by setting the BIDIOE bit in SPI_CR1. It is not enough to change the device driver, this needs to be implemented in the stm32 spi driver.

salkinium, 2017-12-13:
> It is not enough to change the device driver, this needs to be implemented in the stm32 spi driver.

What I meant is that the sensor driver code doesn’t send and receive at the same time anyways. So the half-duplex operation should be transparent to the sensor driver and _only_ exposed to the user as a peripheral driver config option (like setting the baudrate).

> I like the idea of automatically enabling it if the MOSI and MISO pins are the same since it doesn't require modifying the existing interface at all, and it doesn't leave a useless method on other platforms.

Yes, that’s very neat, however the xpcc gpio multiplex API is very broken (see #178 for details and how it’s been fixed in modm), so this won’t easily be doable. Therefore, I recommend the extra API for now as a temp fix.

In modm, this can simply be implemented like so, incl. static compile time tests (ie. if SDIO is used, MISO and MOSI must *not* be used, and vice versa):
```cpp
Spi3::SpiMaster::connect<GpioB5::Sck, GpioB6:Sdio>();
```

salkinium, 2017-12-13:
> Wow now I really feel like an idiot, 

Whoopsie 😛 

> I've literally been writing my own drivers for the LSM303 and LIS3MDL for the last couple of months. I can't believe I missed that xpcc already has a driver for these devices.

Better check how compatible these drivers really are. There are several “almost identical” sensors from ST that then still differ in memory map or transmission protocol. It’s really annoying.